Screenly/Anthias #709: Accept HTTP request from other origin

muex, 2017-12-13:
I wanted to write a client in react and connect to my pi via api. But because of CORS it is not allowed. In my research I found that SimpleHTTPServer has no options to support access controls. Any plans to change that server or any ideas for connecting a js client via api?

vpetersson, 2017-12-14:
We are actually using Flask, which does support CORS. It's rather trivial to add using [Flask-CORS](http://flask-cors.readthedocs.io/en/latest/).

closed 2017-12-15 by muex

vpetersson, 2017-12-15:
@muex we can add this to the next sprint. Alternatively, we're happy to accept a PR.

muex, 2017-12-18:
@vpetersson I closed this issue, because I had an older version on my pi. After your info I updated and I can now work on my ajax frontend. Thank you for this info.