Screenly/Anthias #711: Automate disk image builds

vpetersson, 2017-12-14:
One thing that has been bothering me for a long time is the way we do disk images. It's tedious and error prone. Fortunately, the Raspberry Pi Foundation have made progress when it comes to disk image production (it used to be more or less undocumented). 

There's now a repo called [pi-gen](https://github.com/RPi-Distro/pi-gen) that makes it easy to create a Raspbian image.

I tested it today, and was able to create a Raspbian Lite image with this:

```
$ echo "IMG_NAME='Screenly_OSE'" > config
$ touch ./stage3/SKIP ./stage4/SKIP ./stage5/SKIP
$ rm stage4/EXPORT* stage5/EXPORT*
$ sudo ./build.sh  
```

This is very nice since it also automatically outputs a NOOBS image as well.

What we need to do is basically to create a new 'step' that installs Screenly OSE into this image. That shouldn't be too difficult, but we will likely need to use qemu or similar for this process in order to run things like `ansible` and `pip`.

vpetersson, 2017-12-16:
I cloned the repo and begun the build process. The status can be found in [this branch](https://github.com/Screenly/pi-gen/tree/screenly-ose-lab).

The current issue is that the build process will not install Ansible due to the dependency pyancl. The Screenly specific step can be found [here](https://github.com/Screenly/pi-gen/tree/screenly-ose-lab/stage2/03-screenly).

rusko124, 2018-09-25:
I was able to create an image with these changes https://github.com/rusko124/pi-gen/commit/f8d370cab14b912940e33f3a74e1b7d96015c71d.
For a basis took fork pi-gen with last changes and commits from here https://github.com/Screenly/pi-gen/tree/screenly-ose-lab.

Start build with:
```
$ echo "IMG_NAME='Screenly-OSE'" > config
$ touch ./stage3/SKIP ./stage4/SKIP ./stage5/SKIP
$ rm stage4/EXPORT* stage5/EXPORT*
$ sudo ./build-docker.sh  
```

Results:
```
Image size: 2.71gb
Build time: 101m 52s
```
Result from ansible logs:
```
PLAY RECAP *********************************************************************
localhost                  : ok=61   changed=28   unreachable=0    failed=0 
```

@vpeterson Can you look?

closed 2018-11-19 by vpetersson