Screenly/Anthias #717: URL instead of browser

Kipjr, 2018-01-01:
I've got a virtual machine with screenly-server working. The (custom) display can only view an URL (built-in browser). I would like to visit the screenly-viewer URL but I'm unable to find it. 

While researching viewer.py I only see references to a browser with varying urls.

Can you find a way to let us use URLS instead of the browser?

vpetersson, 2018-01-02:
Hi @Kipjr,

I'm not exactly sure what you're trying to do. The "server" you're referring to is just for development purposes. It's not a full fledged "virtual Screenly OSE" so to speak, as it only has the server components. 

> While researching viewer.py I only see references to a browser with varying urls.

Yes, Screenly viewer talks to [UZBL](https://www.uzbl.org/), which is the browser, over a socket and instructs it what to display.

Kipjr, 2018-01-02:
I wanted to suggest the possibility to not use the UZBL but use an URL which an other device can access. Thks URL should display the same info as the browser

vpetersson, 2018-01-02:
Sorry, but I still don't follow. If you just want to extract the URLs, it would be far easier to just use the [API](http://ose.demo.screenlyapp.com/api/docs/).

Kipjr, 2018-01-02:
> Screenly viewer talks to UZBL, which is the browser, over a socket and instructs it what to display.
Is it possible to let the screenly viewer talk to one PHP-page? This way the limitations of the UZBL-browser are on the client-side and not on the Raspberry Pi

vpetersson, 2018-01-02:
I think there's a misunderstanding here. Screenly OSE is always stand-alone. It always runs on a single Raspberry Pi. The server and client parts are tightly coupled and runs on the same device. 

You can't *just* run the server (at least not in a way that makes any sense). There is a dockerized version of Screenly, but the only reason it exists is to serve the demo page and to help developers. It's not useful for any end-users. 

If what you want to do is to replace the way that the server fires commands to the browser with something else (like an API call), that would in theory be possible, but would probably require a fair amount of work (as that is not what it is designed to do).

closed 2018-01-03 by Kipjr