Screenly/Anthias #720: HTTP basic authentication: not possible with credetials

lazyzero, 2018-01-05:
Hi,

I use the screenly-ose raspberry pi image with the git branch production. I tried to setup the HTTP basic authentication like described in the Wiki, but I am not able to login afterwards. I tried with removing all whitespace on the password and user name line.

vpetersson, 2018-01-05:
Did you restart the service after?

lazyzero, 2018-01-10:
I did it like in the Configuration section of your Wiki 
https://github.com/Screenly/screenly-ose/blob/master/docs/http-basic-authentication.md
after pkill I had to enter the credentials but it failed to authenticate

vpetersson, 2018-01-10:
@antonmolodykh could you try reproducing this?

antonmolodykh, 2018-01-10:
@vpetersson I performed the steps from wiki. It work for me

@lazyzero could you show me the output of `cat /home/pi/.screenly/screenly.conf`?

lazyzero, 2018-01-17:
I could correct the config file by again adding the lines from the wiki with cat.
The default lines that where in the file didn't work for me.

vpetersson, 2018-01-17:
@lazyzero could you post an example of a snippet that works and one that doesn't work?

closed 2018-10-19 by stale[bot]