Screenly/Anthias #721: Unable to interact with the webpage using the keyboard

domlwilliams, 2018-01-05:
I'm unable to interact with the webpage when using the keyboard. I can see the mouse working and know the keyboard is working as it works when the dev tools is open.
I'm trying to view a site which is behind a login and can only think of manually logging in each time the pi is started.

vpetersson, 2018-01-10:
Not sure about this one. The keyboard should work. @antonmolodykh any ideas here?

antonmolodykh, 2018-01-10:
I don't know what's the matter, but I confirm that the keyboard does not work

vpetersson, 2018-01-17:
I want to make this an option that is disabled by default.

putneyj, 2018-02-07:
I am also seeing this issue. Is there some way of re-enabling keyboard input? I am also trying to access sites that are behind a log-in screen.

vpetersson, 2018-02-08:
@antonmolodykh do we have a solution for this?

antonmolodykh, 2018-02-09:
@vpetersson  I have not found a solution for this yet, but I think that it will not matter soon, as we will replace uzbl

vpetersson, 2018-08-13:
The more I think about this, I think this is the wrong approach. A better solution would be to use VNC or similar instead.

jonpfeiler, 2018-08-23:
I am still having an issue with this. Is there any way to enable the keyboard when directly accessing screenly on the pi?

ediphy-azorab, 2018-09-27:
This still appears to be an issue. Any workarounds available?

nv2k, 2018-11-01:
> The more I think about this, I think this is the wrong approach. A better solution would be to use VNC or similar instead.

I'm having the same problem. Tried VNC and it's not working either.
Any thoughts?

closed 2019-01-07 by stale[bot]