Screenly/Anthias #725: Screen stufck in 'Unable to resolve local IP address'

vpetersson, 2018-01-09:
It appears as something in the latest update broke the viewer logic. If you boot a device without any network connection, it will get stuck displaying the splash screen.

vpetersson, 2018-01-14:
I just reproduced this again.

Steps to reproduce (on a Pi3 Model B):

 * Flash out the [latest disk image](https://github.com/Screenly/screenly-ose/releases/download/v0.13/2017-11-25-Screenly_OSE_4GB.zip)
 * Connect the device using Ethernet
 * Optional: Enable SSH
 * Run `./bin/run_upgrade.sh` and install the development release
 * Reboot the device and unplug the Ethernet cable when the four Raspberries appears on the screen.

**Result:** The device just shows "Unable to resolve local IP Address" indefinitely
**Expected result:** The Resin WiFi Connect to appear

vpetersson, 2018-01-15:
So the root cause for this was that `.screenly/wifi_set` was created *prior* to the reboot. This causes the wifi connect never to be launched. There are subsequently two related bugs:
 
* #727
* #728

vpetersson, 2018-01-18:
Unable to reproduce this after the merge.

closed 2018-01-18 by vpetersson

kams01, 2019-04-06:
I recently starting working on this as a user to setup OSE at our facility. I used NOOBS to install OSE. After you move the Pi to another network, the network selection tool does not pop back up after "Unable to resolve IP"

Since then I have used several things - Updated to latest build <- by the way, how do I know which version of OSE am I running? there is no simple way to know this!

Updated to the latest development build (all using SSH or local)
Brought it back to the original network where it attached and hit the "Rerun Network Detection" and it does not reset anything.
Tried to remove the wifi-set file, but it cannot find it.
So basically had to reinstall OSE at the final location and deploy it there - not very practical if you want to test a few things out earlier.

So am I missing anything here?