Screenly/Anthias #726: A couple of minor frontend bugs

darginyegor, 2018-01-10:
**1. Cursor doesn't react on hovering the assets arrangement icon.**
When you move the cursor over the arrangement icon in the assets list, it stays in arrow mode, which is incorrect from the UI point. I even doubted for a second if trying to drag will lead to any result...

**2. Dragging one of the assets doesn't always make other ones jump to correct positions.**
The simple way to reproduce is:
1. Open [Screenly OSE demo page](http://ose.demo.screenlyapp.com/);
2. Drag the first asset down until the second one jumps up and DON'T release it;
3. Move that same asset all the way back up and see the second one doesn't jump back.

vpetersson, 2018-01-10:
Good points, @darginyegor. We'll add this into the next sprint.

darginyegor, 2018-01-10:
**[UPD]** Hovering the link, that says "Get the latest Screenly news..." turns cursor to text mode whaaaat? :slightly_smiling_face:

vpetersson, 2018-08-13:
Solved in some previous fix as i cannot reproduce it on the [demo site](http://ose.demo.screenlyapp.com/).

closed 2018-08-13 by vpetersson