Screenly/Anthias #733: Blank Screen After screenly-ose update

rhodesmi, 2018-01-26:
After upgrading my screenly-ose i now get a blank screen, when i reboot i do see the screenly splash page but thats it.

Here are some logs

Jan 26 07:54:52 raspberrypi python[3713]: Generating asset-list...
Jan 26 07:54:52 raspberrypi python[3713]: Viewer crashed.
Jan 26 07:54:52 raspberrypi python[3713]: Traceback (most recent call last):
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 457, in <module>
Jan 26 07:54:52 raspberrypi python[3713]: main()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 444, in main
Jan 26 07:54:52 raspberrypi python[3713]: scheduler = Scheduler()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 123, in __init__
Jan 26 07:54:52 raspberrypi python[3713]: self.update_playlist()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 167, in update_playlist
Jan 26 07:54:52 raspberrypi python[3713]: (new_assets, new_deadline) = generate_asset_list()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 198, in generate_asset_list
Jan 26 07:54:52 raspberrypi python[3713]: assets = assets_helper.read(db_conn)
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/lib/assets_helper.py", line 90, in read
Jan 26 07:54:52 raspberrypi python[3713]: c.execute(queries.read_all(keys))
Jan 26 07:54:52 raspberrypi python[3713]: OperationalError: no such column: is_processing
Jan 26 07:54:52 raspberrypi python[3713]: Traceback (most recent call last):
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 457, in <module>
Jan 26 07:54:52 raspberrypi python[3713]: main()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 444, in main
Jan 26 07:54:52 raspberrypi python[3713]: scheduler = Scheduler()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 123, in __init__
Jan 26 07:54:52 raspberrypi python[3713]: self.update_playlist()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 167, in update_playlist
Jan 26 07:54:52 raspberrypi python[3713]: (new_assets, new_deadline) = generate_asset_list()
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/viewer.py", line 198, in generate_asset_list
Jan 26 07:54:52 raspberrypi python[3713]: assets = assets_helper.read(db_conn)
Jan 26 07:54:52 raspberrypi python[3713]: File "/home/pi/screenly/lib/assets_helper.py", line 90, in read
Jan 26 07:54:52 raspberrypi python[3713]: c.execute(queries.read_all(keys))
Jan 26 07:54:52 raspberrypi python[3713]: sqlite3.OperationalError: no such column: is_processing
Jan 26 07:54:53 raspberrypi systemd[1]: screenly-viewer.service: main process exited, code=exited, status=1/FAILURE
Jan 26 07:54:53 raspberrypi systemd[1]: Unit screenly-viewer.service entered failed state.
Jan 26 07:54:53 raspberrypi systemd[1]: screenly-viewer.service holdoff time over, scheduling restart.
Jan 26 07:54:53 raspberrypi systemd[1]: Stopping Screenly Viewer...
Jan 26 07:54:53 raspberrypi systemd[1]: Starting Screenly Viewer...
Jan 26 07:54:54 raspberrypi systemd[1]: Started Screenly Viewer.
Jan 26 07:55:04 raspberrypi python[3747]: browser found dead, restarting
Jan 26 07:55:04 raspberrypi python[3747]: Loading browser...
Jan 26 07:55:04 raspberrypi python[3747]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 3759.
Jan 26 07:55:08 raspberrypi python[3747]: current url is file:///tmp/screenly_html/black_page.html
Jan 26 07:55:08 raspberrypi python[3747]: Generating asset-list...
Jan 26 07:55:08 raspberrypi python[3747]: Viewer crashed.
Jan 26 07:55:08 raspberrypi python[3747]: Traceback (most recent call last):
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 457, in <module>
Jan 26 07:55:08 raspberrypi python[3747]: main()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 444, in main
Jan 26 07:55:08 raspberrypi python[3747]: scheduler = Scheduler()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 123, in __init__
Jan 26 07:55:08 raspberrypi python[3747]: self.update_playlist()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 167, in update_playlist
Jan 26 07:55:08 raspberrypi python[3747]: (new_assets, new_deadline) = generate_asset_list()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 198, in generate_asset_list
Jan 26 07:55:08 raspberrypi python[3747]: assets = assets_helper.read(db_conn)
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/lib/assets_helper.py", line 90, in read
Jan 26 07:55:08 raspberrypi python[3747]: c.execute(queries.read_all(keys))
Jan 26 07:55:08 raspberrypi python[3747]: OperationalError: no such column: is_processing
Jan 26 07:55:08 raspberrypi python[3747]: Traceback (most recent call last):
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 457, in <module>
Jan 26 07:55:08 raspberrypi python[3747]: main()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 444, in main
Jan 26 07:55:08 raspberrypi python[3747]: scheduler = Scheduler()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 123, in __init__
Jan 26 07:55:08 raspberrypi python[3747]: self.update_playlist()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 167, in update_playlist
Jan 26 07:55:08 raspberrypi python[3747]: (new_assets, new_deadline) = generate_asset_list()
Jan 26 07:55:08 raspberrypi python[3747]: File "/home/pi/screenly/viewer.py", line 198, in generate_asset_list
[... 486 lines cut ...]
Jan 26 07:58:25 raspberrypi systemd[1]: screenly-viewer.service holdoff time over, scheduling restart.
Jan 26 07:58:25 raspberrypi systemd[1]: Stopping Screenly Viewer...
Jan 26 07:58:25 raspberrypi systemd[1]: Starting Screenly Viewer...
Jan 26 07:58:26 raspberrypi systemd[1]: Started Screenly Viewer.
pi@raspberrypi:~$ 

Also it only appears to be listening on local host on port 8080 so i cannot access via a Browser

pi@raspberrypi:~$ netstat -an | grep -i isten
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN     
tcp6       0      0 :::22                   :::*                    LISTEN    

ps -ef

pi         678     1  0 07:31 ?        00:00:12 /usr/bin/python /home/pi/screenly/server.py
pi         800   678  4 07:32 ?        00:01:22 /usr/bin/python /home/pi/screenly/server.py
pi        5090     1 85 08:05 ?        00:00:05 /usr/bin/python /home/pi/screenly/viewer.py

pi@raspberrypi:~$ sudo systemctl restart screenly*
Failed to restart screenly.service: Unit screenly.service failed to load: No such file or directory.
Failed to restart screenly_assets.service: Unit screenly_assets.service failed to load: No such file or directory.
pi@raspberrypi:~$ 

Any suggestions or shall i just start again from scratch?

vpetersson, 2018-01-26:
@rhodesmi The issue is that the database migration failed.

@antonmolodykh There should be a migration for the new things added with async video processing, right?

antonmolodykh, 2018-01-26:
@vpetersson [of course](https://github.com/Screenly/screenly-ose/blob/master/bin/migrate.py#L167), and as far as I remember it worked fine. I'll check this again and let you know

vpetersson, 2018-01-26:
@antonmolodykh Great.  Thansk.

@rhodesmi in that case, I'd say it's a faulty upgrade. Either the upgrade failed, or it's a faulty SD card (which could cause similar behavior). I'd recommend that you export the data using the export tool inside Screenly and then either replace the SD card or do an overwrite format of the existing SD card using [SD Formatter](https://www.sdcard.org/downloads/formatter_4/).

antonmolodykh, 2018-01-26:
I installed 0.10 version (in which there was no `is_processing` field yet) and successfully updated the system.

vpetersson, 2018-01-26:
Got it, then it's either a fault installation or broken SD card.

closed 2018-01-26 by vpetersson