Screenly/Anthias #735: No DHCP

rhodesmi, 2018-01-26:
Okay so after my failed upgrade i bought a new SD card and installed the latest version - all good so far. Now we run Screenly with no DHCP (i use anx-over to manage it) so i need to set a static IP address (sudo nano /etc/dhcpcd.conf) but SSH is disabled. I was hoping i could enable SSH in the WEBUI but no. So how do i allocate a static IP address? Are we saying i just need to attache a USB keyboard?

vpetersson, 2018-01-26:
Yes, SSH is disabled by default on Raspbian since some time ago. You can find instruction on how to enable it [here](https://www.raspberrypi.org/documentation/remote-access/ssh/) (hint: you don't need a keyboard). There's no way to enable this in Screenly, so you'd need do it either over SSH or using a keyboard.

closed 2018-01-26 by vpetersson

rhodesmi, 2018-01-26:
Thanks for everything I really appreciate your help. I didn’t think it was raspberry that stopped it. Intersestly I was starting to see if it could be done by modifying the sd card directly but would never of guessed that. Thanks again for everything and have a great weekend