Screenly/Anthias #740: Offline mode - assets stop playing after a few days

slbovey, 2018-02-05:
Hello - I'm really sorry to of found an issue with this great software.

I have been testing using a Raspberry Pi 2 with the newest image - the situation i was planning to use it in does not have a network connection - I have installed a hardware clock but after a few days it stops playing assets even after a reboot. (displays 'Screenly' white page that it shows when first booted)
Once connected to the internet it then seems to work again.
I have read that offline should be possible - does it need to check in to a server for updates etc ? - is there a way of disabling this.

Thank you
Simon
p.s. I'm no linux expert but can do the basics - i did look for a log file but couldn't find one in locations i have found on your website \ github entries
p.p.s Sorry if i am missing something easy and wasting you time !

Jir4, 2018-02-06:
Did you check your deadlines ?

slbovey, 2018-02-06:
Yes - before i had the hardware clock i had set the deadlines to the year 3000 as i know the software clock is unpredictable - i've also checked the hardware clock is set right and seems to pickup the right time when logged into the console.

Jir4, 2018-02-06:
You can find some log in /var/log/daemon.log

slbovey, 2018-02-06:
thanks - i'm fairly sure it's trying to get online - when i do connect it up to my network it works fine again (and power cycled - i've tried a power cycle without network and it still doesn't work)

vpetersson, 2018-02-06:
Strange. What's the output you get from `date` when logging into the console?

slbovey, 2018-02-06:
Ok my clock is a few hours out it seems but currently its showing this and working

[image]

i'll leave it on and see if it goes wrong again and then check the clock

vpetersson, 2018-02-06:
Strange indeed. The next step then would be to pull the logs through `journalctl` when the screen isn't displaying anything.

slbovey, 2018-02-06:
Ok thanks - I'll connect it up and wait for it to stop again - just as something to try last night i did connect it up and ran the apt-get update and upgrade commands

I will leave it on and report back - thank you for your attention

slbovey, 2018-02-08:
Hello again - after about a day and a half its stopped again - current time on the clock is 06/02/2018 : 22:05 ish so don't worry that logs are not exactly in date
[log.log](https://github.com/Screenly/screenly-ose/files/1706413/log.log)

I have just restarted it and this time it has started again which it doesn't normally do - i'll see how long it lasts this time

Thanks
Simon

slbovey, 2018-02-09:
it went wrong over night again at some point - similar logs as before - i won't reboot just yet incase you need anything from it - the journalctl looks very similar though

slbovey, 2018-02-14:
I just powered the unit on after being off for a few days - weirdly the 'date' command now returns the correct time - its not been on a network to resync etc
all it's displaying is the 'screenly'  white screen with the expected message about no IP address
Thanks

gitN00bie, 2018-02-22:
Hello,
I seem to be having a similar issue and this is occurring on two different Pi Models and SD cards.  It seems that when I connect the Pi to LAN with internet access, the Screenly OSE app starts working as intended.  Otherwise after a few days connected to a router via ethernet cable but not connected to internet I get the flashing blank/white splash page that flashes every few seconds. No amount of reboots of router or Pi after it boots, just the flashing splash page is what comes up on display.  I should be running the latest image.

antonmolodykh, 2018-02-22:
Hi, @gitN00bie 
I think we have already solved this problem that you described. Pleace run OSE update (development branch).

slbovey, 2018-02-22:
Ah thats great news - wasn't aware of that - do i upgrade using this ?

$ bash <(curl -sL https://www.screenly.io/install-ose.sh)

or how do i specify the development version Please ?

Thank you

antonmolodykh, 2018-02-22:
@slbovey  all right. run `bash <(curl -sL https://www.screenly.io/install-ose.sh)` and follow the instructions, you will be asked to install the development version

slbovey, 2018-02-22:
Thanks antonmolodykh - will give it a try later when i can connect it up again :)

slbovey, 2018-02-26:
Updated Thursdsay evening - put back into action friday - but at some point over the weekend its stopped again - back to the same issue i'm afraid - i did choose the development version.

vpetersson, 2018-02-26:
I have a feeling this is related to #744.

slbovey, 2018-02-26:
Ok thanks - looks like better brains than mine are on the issue :)
For detail clarification (incase it helps) I'm running a Pi2 (offline - no network connection) with a hardware clock - i did try a power cycle this morning but it made no difference.

antonmolodykh, 2018-02-26:
@slbovey @gitN00bie I made a fix for this, please run the update (development branch)

slbovey, 2018-02-26:
Thanks @antonmolodykh - will give it a go tonight

closed 2018-10-19 by stale[bot]