Screenly/Anthias #741: Screenly Viewer crashes after a few hours without internet connection

Andreas879, 2018-02-06:
First of all, thank you for developing Screenly. It's just great.
I have installed the latest Screenly OSE 20180119 Image. Also in the previous version, after a few hours the Screenly Viewer crashes. Screenly runs in a network without internet connection. NTP service is installed and running too. As soon as Screenly can access to the internet, the viewer works again. I can reproduce this error by blocking the Internet access. I think it is because of the missing update check

syslog

Feb  6 17:12:37 Infoboard systemd[1]: Started Screenly Viewer.
Feb  6 17:12:38 Infoboard systemd[1]: Started Fail2Ban Service.
Feb  6 17:12:38 Infoboard systemd[1]: Reached target Multi-User System.
Feb  6 17:12:38 Infoboard systemd[1]: Reached target Graphical Interface.
Feb  6 17:12:38 Infoboard systemd[1]: Starting Update UTMP about System Runlevel Changes...
Feb  6 17:12:38 Infoboard systemd[1]: Started Update UTMP about System Runlevel Changes.
Feb  6 17:12:38 Infoboard systemd[1]: Startup finished in 1.477s (kernel) + 10.164s (userspace) = 11.642s.
Feb  6 17:12:39 Infoboard python[540]: [2018-02-06 17:12:39 +0000] [540] [INFO] Starting gunicorn 19.4.5
Feb  6 17:12:39 Infoboard python[540]: [2018-02-06 17:12:39 +0000] [540] [INFO] Listening at: http://127.0.0.1:8080 (540)
Feb  6 17:12:39 Infoboard python[540]: [2018-02-06 17:12:39 +0000] [540] [INFO] Using worker: threads
Feb  6 17:12:39 Infoboard python[540]: [2018-02-06 17:12:39 +0000] [622] [INFO] Booting worker with pid: 622
Feb  6 17:12:39 Infoboard python[568]: browser found dead, restarting
Feb  6 17:12:39 Infoboard python[568]: Loading browser...
Feb  6 17:12:39 Infoboard python[568]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 626.
Feb  6 17:12:48 Infoboard systemd[1]: Time has been changed
Feb  6 17:12:48 Infoboard systemd[1]: apt-daily.timer: Adding 5h 56min 29.531389s random time.
Feb  6 17:12:48 Infoboard systemd[1]: apt-daily-upgrade.timer: Adding 11min 53.469582s random time.
Feb  6 17:12:48 Infoboard python[568]: current url is file:///tmp/screenly_html/black_page.html
Feb  6 17:12:48 Infoboard python[568]: Generating asset-list...
Feb  6 17:12:48 Infoboard python[568]: Viewer crashed.
Feb  6 17:12:48 Infoboard python[568]: Traceback (most recent call last):
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
Feb  6 17:12:48 Infoboard python[568]:     main()
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 452, in main
Feb  6 17:12:48 Infoboard python[568]:     asset_loop(scheduler)
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 373, in asset_loop
Feb  6 17:12:48 Infoboard python[568]:     check_update()
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 343, in check_update
Feb  6 17:12:48 Infoboard python[568]:     'Hash': str(git_hash),
Feb  6 17:12:48 Infoboard python[568]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 95, in track
Feb  6 17:12:48 Infoboard python[568]:     self._consumer.send('events', json_dumps(event, cls=self._serializer))
Feb  6 17:12:48 Infoboard python[568]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 353, in send
Feb  6 17:12:48 Infoboard python[568]:     self._write_request(self._endpoints[endpoint], json_message, api_key)
Feb  6 17:12:48 Infoboard python[568]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 376, in _writ                                                              e_request
Feb  6 17:12:48 Infoboard python[568]:     six.raise_from(MixpanelException(e), e)
Feb  6 17:12:48 Infoboard python[568]: AttributeError: 'module' object has no attribute 'raise_from'
Feb  6 17:12:48 Infoboard python[568]: Traceback (most recent call last):
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
Feb  6 17:12:48 Infoboard python[568]:     main()
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 452, in main
Feb  6 17:12:48 Infoboard python[568]:     asset_loop(scheduler)
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 373, in asset_loop
Feb  6 17:12:48 Infoboard python[568]:     check_update()
Feb  6 17:12:48 Infoboard python[568]:   File "/home/pi/screenly/viewer.py", line 343, in check_update
Feb  6 17:12:48 Infoboard python[568]:     'Hash': str(git_hash),
Feb  6 17:12:48 Infoboard python[568]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 95, in track
Feb  6 17:12:48 Infoboard python[568]:     self._consumer.send('events', json_dumps(event, cls=self._serializer))
Feb  6 17:12:48 Infoboard python[568]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 353, in send
Feb  6 17:12:48 Infoboard python[568]:     self._write_request(self._endpoints[endpoint], json_message, api_key)
Feb  6 17:12:48 Infoboard python[568]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 376, in _writ                                                              e_request
Feb  6 17:12:48 Infoboard python[568]:     six.raise_from(MixpanelException(e), e)
Feb  6 17:12:48 Infoboard python[568]: AttributeError: 'module' object has no attribute 'raise_from'
Feb  6 17:12:48 Infoboard systemd[1]: screenly-viewer.service: Main process exited, code=exited, status=1/FAILURE
Feb  6 17:12:48 Infoboard systemd[1]: screenly-viewer.service: Unit entered failed state.
Feb  6 17:12:48 Infoboard systemd[1]: screenly-viewer.service: Failed with result 'exit-code'.
Feb  6 17:12:49 Infoboard systemd[1]: screenly-viewer.service: Service hold-off time over, scheduling restart.
Feb  6 17:12:49 Infoboard systemd[1]: Stopped Screenly Viewer.
Feb  6 17:12:49 Infoboard systemd[1]: Starting Screenly Viewer...
Feb  6 17:12:49 Infoboard systemd[1]: Started Screenly Viewer.

vpetersson, 2018-02-06:
Good find, @Andreas879. @antonmolodykh can you take a look?

closed 2018-02-07 by vpetersson

antonmolodykh, 2018-02-07:
Thanks for your find, @Andreas879 
I made a fix, please run update (development version)

Andreas879, 2018-02-07:
Thanks for the fast bugfix. Update already installed.

collinbarlage, 2018-06-13:
Is it possible to access the shell at this screen? ctrl-shift-F1 doesn't work