Screenly/Anthias #743: Initial Resin WiFi Connect screen not fitting screen

metalgeekstl, 2018-02-07:
Hardware: Raspberry Pi Zero W

At first boot, I'm able to see the SSID on my phone created by Resin WiFi Connect, but it's not scaled correctly and I can't see what the passphrase is. 
[image]

I've tried 3 different displays. All have the same issue.

vpetersson, 2018-02-07:
What resolution screen are you using?

metalgeekstl, 2018-02-07:
I have no way of knowing. Plug in TV, plug in Pi, that's what displays on each of the 3 different TVs I've tried.

vpetersson, 2018-02-07:
But all the monitors are capable of doing 1080p? That is the assumption that Screenly makes.

metalgeekstl, 2018-02-07:
They should be. Standard TVs. Maybe the Pi isn't sending the TV the correct resolution?

metalgeekstl, 2018-02-07:
Is there another way to find the passphrase? If I had that I'd be in business.

vpetersson, 2018-02-07:
you are likely able to scroll using a keyboard. However, I'd try to swap out the HDMI cable (and/or) adaptor as that may cause the issue.

metalgeekstl, 2018-02-07:
Keyboard or mouse does nothing. I've used different HDMI cables for each TV, and I don't have any other micro HDMI adapter available. This Pi has never had any display resolution issues outside of Screenly OSE. That's the common denominator.

vpetersson, 2018-02-07:
Screenly doesn't really do anything with the resolution. This is all inherited by Raspbian. As such, there is no Screenly setting to change this. What happens is that in the early boot process, the Raspberry Pi will "negotiate" with the monitor for the supported resolution(s). Adaptors can interfere and break this negotiation. 

There are ways that you can override this through `config.txt` and you can find more documentation about that [here](https://elinux.org/RPiconfig).

vpetersson, 2018-02-20:
Closing ticket due to lack of response.

closed 2018-02-20 by vpetersson

pletiplot, 2019-11-06:
Try connect the raspberry to an ethernet connection and then you can try more things.