Screenly/Anthias #744: Viewer crashed

ddn, 2018-02-10:
I have 2 screenly installs both running latest release (014bc04...) that are in an infinite loop of crashing, about once per minute.

I mentioned in a closed thread that I went looking for logs because of a crashing issue, my older rpi2 dies completely eventually, and my pi3 gets stuck on the IP screen and also needs a reboot.

Feb  4 19:36:28 raspberrypi python[6057]: Viewer crashed.
Feb  4 19:36:28 raspberrypi python[6057]: Traceback (most recent call last):
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
Feb  4 19:36:28 raspberrypi python[6057]:     main()
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 452, in main
Feb  4 19:36:28 raspberrypi python[6057]:     asset_loop(scheduler)
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 373, in asset_loop
Feb  4 19:36:28 raspberrypi python[6057]:     check_update()
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 343, in check_update
Feb  4 19:36:28 raspberrypi python[6057]:     'Hash': str(git_hash),
Feb  4 19:36:28 raspberrypi python[6057]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 95, in track
Feb  4 19:36:28 raspberrypi python[6057]:     self._consumer.send('events', json_dumps(event, cls=self._serializer))
Feb  4 19:36:28 raspberrypi python[6057]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 353, in send
Feb  4 19:36:28 raspberrypi python[6057]:     self._write_request(self._endpoints[endpoint], json_message, api_key)
Feb  4 19:36:28 raspberrypi python[6057]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 376, in _write_request
Feb  4 19:36:28 raspberrypi python[6057]:     six.raise_from(MixpanelException(e), e)
Feb  4 19:36:28 raspberrypi python[6057]: AttributeError: 'module' object has no attribute 'raise_from'
Feb  4 19:36:28 raspberrypi python[6057]: Traceback (most recent call last):
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
Feb  4 19:36:28 raspberrypi python[6057]:     main()
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 452, in main
Feb  4 19:36:28 raspberrypi python[6057]:     asset_loop(scheduler)
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 373, in asset_loop
Feb  4 19:36:28 raspberrypi python[6057]:     check_update()
Feb  4 19:36:28 raspberrypi python[6057]:   File "/home/pi/screenly/viewer.py", line 343, in check_update
Feb  4 19:36:28 raspberrypi python[6057]:     'Hash': str(git_hash),
Feb  4 19:36:28 raspberrypi python[6057]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 95, in track
Feb  4 19:36:28 raspberrypi python[6057]:     self._consumer.send('events', json_dumps(event, cls=self._serializer))
Feb  4 19:36:28 raspberrypi python[6057]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 353, in send
Feb  4 19:36:28 raspberrypi python[6057]:     self._write_request(self._endpoints[endpoint], json_message, api_key)
Feb  4 19:36:28 raspberrypi python[6057]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 376, in _write_request
Feb  4 19:36:28 raspberrypi python[6057]:     six.raise_from(MixpanelException(e), e)
Feb  4 19:36:28 raspberrypi python[6057]: AttributeError: 'module' object has no attribute 'raise_from'
Feb  4 19:36:28 raspberrypi python[6057]: Exception in thread Thread-3 (most likely raised during interpreter shutdown):Exception in thread Thread-4 (most likely raised during interpreter shutdown):
Feb  4 19:36:28 raspberrypi python[6057]: Traceback (most recent call last):
Feb  4 19:36:28 raspberrypi systemd[1]: screenly-viewer.service: Main process exited, code=exited, status=1/FAILURE
Feb  4 19:36:28 raspberrypi systemd[1]: screenly-viewer.service: Unit entered failed state.
Feb  4 19:36:28 raspberrypi systemd[1]: screenly-viewer.service: Failed with result 'exit-code'.
Feb  4 19:36:28 raspberrypi systemd[1]: screenly-viewer.service: Service hold-off time over, scheduling restart.
Feb  4 19:36:28 raspberrypi systemd[1]: Stopped Screenly Viewer.
Feb  4 19:36:28 raspberrypi systemd[1]: Starting Screenly Viewer...
Feb  4 19:36:29 raspberrypi systemd[1]: Started Screenly Viewer.
Feb  4 19:36:31 raspberrypi python[11246]: browser found dead, restarting
Feb  4 19:36:31 raspberrypi python[11246]: Loading browser...
Feb  4 19:36:31 raspberrypi python[11246]: Browser loading http://127.0.0.1:8080/splash_page. Running as PID 11261.
Feb  4 19:36:33 raspberrypi python[11246]: current url is http://127.0.0.1:8080/splash_page
Feb  4 19:37:34 raspberrypi python[11246]: Scheduler init
Feb  4 19:37:34 raspberrypi python[11246]: update_playlist
Feb  4 19:37:34 raspberrypi python[11246]: Generating asset-list...
Feb  4 19:37:34 raspberrypi python[11246]: generate_asset_list deadline: 1970-01-19 16:07:00
Feb  4 19:37:34 raspberrypi python[11246]: update_playlist done, count 6, counter 0, index 0, deadline 1970-01-19 16:07:00
Feb  4 19:37:34 raspberrypi python[11246]: Entering infinite loop.

vpetersson, 2018-02-11:
Are you running in offline mode? If so, this has been solved in master, so please switch to the development branch and it should hopefully be resolved.

ddn, 2018-02-11:
@vpetersson negative. They both have a 24/7 wifi connection. One thing I can add is that I've noticed this loop doesn't begin immediately, something starts it after it runs fine for a while, but then once it loops crashing eventually the machine dies completely.

philpalnau, 2018-02-13:
Any update on this? I just recently updated to the development branch since I was having the same exact issue. Ill update back if that fixes the issue, is there an ETA for pushing this fix into Prod?

Thanks!

vpetersson, 2018-02-13:
Do we have any lead on this @antonmolodykh?

@philpalnau i'd suggest rolling back to an earlier release as a temporary measure as we don't have a fix ready for it yet.

antonmolodykh, 2018-02-16:
@ddn @philpalnau  please run the update from master branch, I made the fix. let me know if there are problems again, thank you.

philpalnau, 2018-02-16:
I updated from a fresh install. I will update on the results. Thanks for
the fast replies! You guys rock

ddn, 2018-02-19:
Updated stable & development branch units. Will circle back later in the week and report results.

ddn, 2018-02-20:
Behavior is unchanged on master. Still lasts about 24 hours before crashing for good. It goes maybe 8 hours before the Viewer crashed starts once per minute:

Feb 20 03:54:09 raspberrypi python[20350]: Viewer crashed.
Feb 20 03:55:32 raspberrypi python[20405]: Viewer crashed.
Feb 20 03:56:55 raspberrypi python[20459]: Viewer crashed.

As an update, I was unclear in my previous post. I updated the pi running master, and I backported the fix to the older pi running production to see if that worked there.

However, the older pi (running production) does seem to be stable now with the wifi power management disabled.

Let me know how I can help debug.

vpetersson, 2018-02-20:
:( 

@antonmolodykh I suggest you fork out and create a new branch with more logging.

ddn, 2018-02-20:
I just added munin to the pi running master, I have a hunch from some of the log messages that there might be a memory leak involved. Not sure if it's the cause or effect. I'll keep an eye on that.

vpetersson, 2018-02-20:
Good idea, @ddn

antonmolodykh, 2018-02-21:
@ddn okay, I'll figure it out. let's just in case, clarify what is happening now: you have RPi3, on which you updated the OSE (development version). but after about 24 hours the viewer breaks again. The device is connected via wifi. all right?

ddn, 2018-02-21:
Correct.

I'dd add 2 things since I started logging with munin.

Memory usage on the rpi3 on master climbs over a period of about 6 hours until the load average shoots through the roof and the system effectively locks up. It looks like it takes about an hour to come back.

The rpi2 on production seems to be stable with the power management fix, but it's a little too early to say for sure.

vpetersson, 2018-02-21:
> Memory usage on the rpi3 on master climbs over a period of about 6 hours until the load average shoots through the roof and the system effectively locks up. It looks like it takes about an hour to come back.

Interesting. What content is being displayed? Is it heavy web pages? If so, it's likely a memory leak in `uzbl`.

ddn, 2018-02-21:
On that one, yeah. I'm running a geckoboard dashboard that is probably more JS heavy than it should be.

vpetersson, 2018-02-21:
Ah yeah, that's def on the heavier side. What duration do you have set? In theory it should release the memory whenever reloaded. If not, adding an image or something between should force that.

ddn, 2018-02-21:
Just to be super accurate, two 30-second instances of the same URL
(separated by an image), 2 static images, one mp4, and one screenly weather
widget.

vpetersson, 2018-02-21:
Yeah let's give that a go to see if that help.

ddn, 2018-02-22:
I am in fact still seeing the Viewer crashed loop on production with the wifi power management fix backported.

As for the master branch one running out of memory, it's definitely uzbl which is apparently not releasing memory. Where would you suggest adding a full restart of the uzbl instance?

This may be a fluke or some sort of race condition, but after one time the system ran out of memory it did come back, but network-manager was using 100% CPU until it finally crashed completely.

vpetersson, 2018-02-23:
Super interesting find about network-manager. That would indeed explain why it started happening in this release. @antonmolodykh can you take a look at this?

ddn, 2018-02-25:
@vpetersson @antonmolodykh Is there a fix available for the original problem of the Viewer crashed loop, to get it out of the crashing loop without rebooting?

I've been able to at least get the systems stable by just killing uzbl once an hour, but once it enters the crash loop I don't see any avenue for a fix other than a reboot.

This is getting pretty frustrating, we've used screenly for years and years and it has typically been very stable but now it requires sometimes twice-daily maintenance to make sure our screens aren't flickering or just black.

vpetersson, 2018-02-25:
@ddn You have a few options:

 * You can have a script that that restart network-manager and kills uzbl (`pkill -f uzbl-core` should do)
 * You could roll back to an earlier version while we try to get to the bottom of this

antonmolodykh, 2018-02-26:
@ddn  I made a fix for this, please run the update (development branch)

ddn, 2018-02-26:
Awesome, going to give that a shot. Thanks @antonmolodykh 

Regarding the memory leak, I haven't done anything with network manager, but killing uzbl-core on the hour has made the system memory-stable. I figure worst case I can back that time off to a greater interval.

vpetersson, 2018-02-27:
@ddn - I'm happy to report that we're making good progress on replacing UZBL (which will solve numerous issues that have been hunting us for some time).

ddn, 2018-03-02:
@vpetersson 👍 

And I'm happy to report both my systems have been stable this week. Still have the cron job killing uzbl regularly, but that I can live with.

Thanks guys!

closed 2018-03-02 by ddn

vpetersson, 2018-03-02:
@ddn Thanks for getting back. How frequently do you need to kill uzbl?

ddn, 2018-03-02:
I've had it on the hour this week. Just backed it off to every 2 hours to see what that does. Judging by the memory usage from munin I could probably go up to every 6 hours and it would be fine, but the distruption of killing it is negligible so I don't have any reason to push it.

ddn, 2018-03-02:
I think you could do something more elegant very easily without introducing a lag or anything. Look for any asset other than a URL, greater than a few seconds in duration, and take that as an opportunity to kill uzbl once per asset loop.

Can you see any issues with that?