Screenly/Anthias #746: Can't configure static IP

abrenoch, 2018-02-16:
I'm not sure what has changed, but trying to configure the network settings in the most recent update is, I'm sorry to say, an _absolute_ mess.

I look at an article barely 3 months old (https://support.screenly.io/hc/en-us/articles/213678343-How-do-I-configure-a-Screenly-node-to-use-a-static-IP-) that says to use the config tool. When following the link to the tool one of the first things it says is "This tool can no longer be used with Screenly OSE", and provides a link to this page (https://www.screenly.io/blog/2018/01/19/screenly-ose-0.13-is-available/) which says to edit `/etc/network/interfaces`. Okay, so I open that file and one of the first things that file says is "For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'". So I hop into that file and make the needed changes, reboot, and no change to my IP... So I go back a step and try modifying `/etc/network/interfaces` like the old way, reboot, and network connectivity does not work at all. Okay, so I go back to the page I first linked to to follow the manual process - which tells me to do exactly  what the second page I linked to says no longer works.

Frankly, I can't figure out how to set a static IP since the 'paradise city' update... and all the instructions I have found lead me in circles.

Can we please get some clear instructions on how to manually modify the network settings? Nothing seems to be working for me now.

Thanks!

vpetersson, 2018-02-16:
Let's see if we can get to the bottom of this.

First, are you trying configure WiFi or Ethernet?

abrenoch, 2018-02-16:
Thanks for the quick reply!

Connected to ethernet - it is an rpi3 so it does have built-in wifi.

antonmolodykh, 2018-02-19:
Hi, @abrenoch 
Some time ago we stopped using our screenly-net-manager for OSE, so now the network configured with standart Raspbian tools.
To configure a static IP for an ethernet-connection, you can simply add this to the `/etc/network/interfaces` file:
```
auto eth0
iface eth0 inet static
        address <ip>
        netmask <netmask>
        gateway <gateway>
```
and reboot the device.

glad to help

vpetersson, 2018-02-20:
Closing out the issue. Will re-open if it doesn't solve the problem.

closed 2018-02-20 by vpetersson

abrenoch, 2018-02-20:
thanks @antonmolodykh that did seem to work, which is strange because I copied my previous settings (from an old screenly install) from that file verbatim and was not able to get onto the network.. I guess the 'edit dhcpcd.conf' message in that interfaces file was totally moot.

I feel like the ability to set the settings with 'network.ini' was a bit of an easier solution, but I'm sure you have your reasons for changing it!

Thanks you for the help guys!

fritserasmus, 2018-03-03:
@vpetersson Thanks for your answer.
Allow me to explain it is not working for me:

I downloaded the latest Screenly OSE ver on 2018-03-01 and installed on my Raspberry Pi 3 Model B

I was planning to deploy the Rpi with Screenly OSE in a location with only WiFi connection.

The configuration using the captive portal worked a charm.

On reboot it was easy to see the IP addres to access screenly assets from another PC. No issues here.

Well here is the issue:
If screenly have been running overnight and I try to access screenly via the webbrowser it would not connect. No power interruptions.

My DHCP server also does not report the Rpi as connected, so I cannot verify the IP Address either

I plugged in a keyboard and ctrl+alt+F1 to confirm th eIP address. Shure the address was still the same and the pc was still running, but could not access the webpage like previous evening.

My only option then would be to reboot to let it reconnect to the DHCP server.
Surprise: It gets a new IP address from the DHCP server even though the lease period has not yet lapsed. 
I am unable to identify the IP Address from the router as I have three other Raspberry Pi's connected. (Doing other tasks PV Monitoring)

So I have two options: 
Either leave the splash screen on and reboot to see the IP address and then connect and risk the integrity of Screenly as anyne with access to the given network can reboot and access screenly via webinterface.:-(

OR assign a static IP address and do not show the splash screen

So I followed the instructions provided: https://github.com/Screenly/screenly-ose/issues/746

Mine:

auto wlan0 
iface wlan0 inet static
        address 192.168.1.115
        netmask 255.255.255.0
        gateway 192.168.1.1

So I was exited to find a detailed instruction for th eversion I am using.

Reboot with sudu reboot
Yet a bigger surprise: Neither putty NOR webiterface can access this ip address!

ctr+alt+F1 to check the IP address: Yes it is "192.168.1.115"
My router does not confirm connection and I am unable to connect to it.

Back to square 0

console, comment out the wlan0 setting in the /etc/network/interfaces file:

So for now my ONLY option to add, remove or change assets is to leave splash screen ON, reboot so I can see the IP address and to access screenly web interface.

Any pointers?

antonmolodykh, 2018-03-12:
Hi @fritserasmus 
If I correctly understood your problem, then this is not related to a static/dynamic ip. this is related to the wifi power management in the latest versions of Raspbian Stretch. We made an update for this, so please run the OSE update (development version) and let me know if there are any problems

vpetersson, 2018-03-13:
Hi @fritserasmus 

I think there is some confusion here, which is understandable. Let me take a step back to explain what I think is the root cause for at least the network config (@antonmolodykh's point above should hopefully solve the other one).

* There ware two (major) ways to configure networking in Ubuntu/Debian/Raspbian:
  * `/etc/network/interfaces`
  * Network Manager

The former is the older method, that is still widely used. When you configure networking using the captive portal, this will configure the wireless network using Network Manager. This means that if you *also* try to configure the same network interface using the other method, you're likely going to end up having trouble.

It is certainly possible to configure the wifi using `etc/network/interfaces`, but the stanza would look something like this then:

```
auto wlan0
iface wlan0 inet static
        address <ip>
        netmask <netmask>
        gateway <gateway>
        wpa-ssid <your SSID>
        wpa-psk <your passphrase>
```

To my knowledge, you cannot mix and match between Network Manager by setting just a static IP in the interfaces file.

If you instead want to use purely use Network Manager, I recommend looking into `nmcli` (or `nmtui` if you prefer a graphical interface).

Hope that helps.

gady87, 2019-11-13:
> I'm not sure what has changed, but trying to configure the network settings in the most recent update is, I'm sorry to say, an _absolute_ mess.
> 
> I look at an article barely 3 months old (https://support.screenly.io/hc/en-us/articles/213678343-How-do-I-configure-a-Screenly-node-to-use-a-static-IP-) that says to use the config tool. When following the link to the tool one of the first things it says is "This tool can no longer be used with Screenly OSE", and provides a link to this page (https://www.screenly.io/blog/2018/01/19/screenly-ose-0.13-is-available/) which says to edit `/etc/network/interfaces`. Okay, so I open that file and one of the first things that file says is "For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'". So I hop into that file and make the needed changes, reboot, and no change to my IP... So I go back a step and try modifying `/etc/network/interfaces` like the old way, reboot, and network connectivity does not work at all. Okay, so I go back to the page I first linked to to follow the manual process - which tells me to do exactly what the second page I linked to says no longer works (making a network.ini file in boot).
> 
> Frankly, I can't figure out how to set a static IP since the 'paradise city' update... and all the instructions I have found lead me in circles.
> 
> Can we please get some clear instructions on how to manually modify the network settings? Nothing seems to be working for me now.
> 
> Thanks!

hi i have not been able to open that file i dont see it in any folder. how you manage to open it ? i tried to connect thru ssh and it says connection refused.

vpetersson, 2019-11-14:
@gady87 Yes, that's because OpenSSH is disabled by default in Rasbian (and subsequently Screenly OSE). You need to enable it first using `raspi-config`.

CodeBrauer, 2021-07-08:
In case anyone searching how to set up the static ip for the ethernet (eth0) connection: You have to use `nmcli` like [described here](https://www.screenly.io/blog/2018/01/19/screenly-ose-0.13-is-available/#:~:text=For%20custom%20Ethernet%20configuration%20and%20advanced%20configurations%2C%20you%20can%20either%20use%20nmcli%20by%20hand%20on%20the%20device%2C%20or%20configure%20networking%20through%20/etc/network/interfaces.)

_Tested with the current release screenly ose 10.1 (based on Raspbian 9.11)_

Basically, you just enable ssh - connect and run this commands with your settings:

```
$ nmcli dev status
DEVICE  TYPE      STATE         CONNECTION
eth0    ethernet  connected     Wired connection 1
wlan0   wifi      disconnected  --
lo      loopback  unmanaged     --

$ sudo nmcli con mod 'Wired connection 1' ipv4.addresses 10.0.3.10/8
$ sudo nmcli con mod 'Wired connection 1' ipv4.gateway 10.0.0.1
$ sudo nmcli con mod 'Wired connection 1' ipv4.dns 10.0.0.1
$ sudo nmcli con mod 'Wired connection 1' ipv4.method manual
$ sudo nmcli con 'Wired connection 1' up
```
Now your ssh session should be broken (Like this: `packet_write_wait: Connection to 10.200.0.99 port 22: Broken pipe`)

And your screenly-ose should have a static ip on eth0 from now on.

gady87, 2021-07-08:
does this also work with the wifi ?

CodeBrauer, 2021-07-09:
@gady87 I think it should, but better make a fresh installation on a test-device in case things break.