Screenly/Anthias #749: Weather webpage not loading correctly

justinkruit, 2018-02-21:
Hello,

I've made my own page to display the weather to use in Screenly. Here's a link: https://cdn.justinkruit.me/jkmonitors/weather.html

Although, when I load this into Screenly, this is the result: [image]

It works absolutely fine on PC though, and the logs are saying nothing about it. Is there something wrong with the player so it won't allow specific code or something?

vpetersson, 2018-02-21:
Yo might be using HTML/CSS that isn't supported by the browser. The current browser is a bit old (and we're working at replacing it with a more modern one).

vpetersson, 2018-08-13:
@justinkruit did you try the new experimental branch?

closed 2018-10-19 by stale[bot]