Screenly/Anthias #754: Unable to load page (black_page.html)

edocastri, 2018-03-01:
Hi!

I was using Screenly OSE whitour problems for almost a month at my office waiting room, but suddenly it's showing this error:
**Unable to load page**
Problem occurred while loading the URL
file ///tmp/screenly_html/black_page.html
Error opening file ///tmp/screenly_html/black_page.html No such device or address.

At this time I'm dealing with this just rebooting the RP through SSH, but I would like to know what's failing to stop worrying about this, any hint would help :)

[image]

vpetersson, 2018-03-01:
Hard to say without any logs. What's the output of `journalctl`?

edocastri, 2018-03-01:
Sorry for the long log, hope it helps:

Logs begin at Thu 2016-11-03 14:16:42 -03, end at Thu 2018-03-01 10:52:42 -03. --
nov 03 14:16:42 rsp_tvventas1 kernel: Booting Linux on physical CPU 0x0
nov 03 14:16:42 rsp_tvventas1 kernel: Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG cro
nov 03 14:16:42 rsp_tvventas1 kernel: CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
nov 03 14:16:42 rsp_tvventas1 kernel: CPU: div instructions available: patching division code
nov 03 14:16:42 rsp_tvventas1 kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
nov 03 14:16:42 rsp_tvventas1 kernel: OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
nov 03 14:16:42 rsp_tvventas1 kernel: cma: Reserved 8 MiB at 0x33800000
nov 03 14:16:42 rsp_tvventas1 kernel: Memory policy: Data cache writealloc
nov 03 14:16:42 rsp_tvventas1 kernel: On node 0 totalpages: 212992
nov 03 14:16:42 rsp_tvventas1 kernel: free_area_init_node: node 0, pgdat 80c6f380, node_mem_map b30a6000
nov 03 14:16:42 rsp_tvventas1 kernel:   Normal zone: 1872 pages used for memmap
nov 03 14:16:42 rsp_tvventas1 kernel:   Normal zone: 0 pages reserved
nov 03 14:16:42 rsp_tvventas1 kernel:   Normal zone: 212992 pages, LIFO batch:31
nov 03 14:16:42 rsp_tvventas1 kernel: percpu: Embedded 14 pages/cpu @b3060000 s25600 r8192 d23552 u57344
nov 03 14:16:42 rsp_tvventas1 kernel: pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
nov 03 14:16:42 rsp_tvventas1 kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
nov 03 14:16:42 rsp_tvventas1 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 211120
nov 03 14:16:42 rsp_tvventas1 kernel: Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=9
nov 03 14:16:42 rsp_tvventas1 kernel: PID hash table entries: 4096 (order: 2, 16384 bytes)
nov 03 14:16:42 rsp_tvventas1 kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
nov 03 14:16:42 rsp_tvventas1 kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
nov 03 14:16:42 rsp_tvventas1 kernel: Memory: 822624K/851968K available (7168K kernel code, 486K rwdata, 2012K rodata, 1
nov 03 14:16:42 rsp_tvventas1 kernel: Virtual kernel memory layout:
                                          vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                          fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                          vmalloc : 0xb4800000 - 0xff800000   (1200 MB)
                                          lowmem  : 0x80000000 - 0xb4000000   ( 832 MB)
                                          modules : 0x7f000000 - 0x80000000   (  16 MB)
                                            .text : 0x80008000 - 0x80800000   (8160 kB)
                                            .init : 0x80b00000 - 0x80c00000   (1024 kB)
                                            .data : 0x80c00000 - 0x80c79834   ( 487 kB)
                                             .bss : 0x80c7b000 - 0x80d3b964   ( 771 kB)
nov 03 14:16:42 rsp_tvventas1 kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
nov 03 14:16:42 rsp_tvventas1 kernel: Hierarchical RCU implementation.
nov 03 14:16:42 rsp_tvventas1 kernel:         Build-time adjustment of leaf fanout to 32.
nov 03 14:16:42 rsp_tvventas1 kernel: NR_IRQS:16 nr_irqs:16 16
nov 03 14:16:42 rsp_tvventas1 kernel: arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
nov 03 14:16:42 rsp_tvventas1 kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max
nov 03 14:16:42 rsp_tvventas1 kernel: sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
nov 03 14:16:42 rsp_tvventas1 kernel: Switching to timer-based delay loop, resolution 52ns
nov 03 14:16:42 rsp_tvventas1 kernel: Console: colour dummy device 80x30
nov 03 14:16:42 rsp_tvventas1 kernel: console [tty1] enabled
nov 03 14:16:42 rsp_tvventas1 kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 B
nov 03 14:16:42 rsp_tvventas1 kernel: pid_max: default: 32768 minimum: 301
nov 03 14:16:42 rsp_tvventas1 kernel: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
nov 03 14:16:42 rsp_tvventas1 kernel: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
nov 03 14:16:42 rsp_tvventas1 kernel: Disabling memory control group subsystem
nov 03 14:16:42 rsp_tvventas1 kernel: CPU: Testing write buffer coherency: ok
nov 03 14:16:42 rsp_tvventas1 kernel: ftrace: allocating 22392 entries in 66 pages
nov 03 14:16:42 rsp_tvventas1 kernel: CPU0: update cpu_capacity 1024
nov 03 14:16:42 rsp_tvventas1 kernel: CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
nov 03 14:16:42 rsp_tvventas1 kernel: Setting up static identity map for 0x100000 - 0x100034
nov 03 14:16:42 rsp_tvventas1 kernel: CPU1: update cpu_capacity 1024
nov 03 14:16:42 rsp_tvventas1 kernel: CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
nov 03 14:16:42 rsp_tvventas1 kernel: CPU2: update cpu_capacity 1024
nov 03 14:16:42 rsp_tvventas1 kernel: CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
nov 03 14:16:42 rsp_tvventas1 kernel: CPU3: update cpu_capacity 1024
nov 03 14:16:42 rsp_tvventas1 kernel: CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
nov 03 14:16:42 rsp_tvventas1 kernel: Brought up 4 CPUs
nov 03 14:16:42 rsp_tvventas1 kernel: SMP: Total of 4 processors activated (153.60 BogoMIPS).
[... cut ...]
mar 01 10:52:42 rsp_tvventas1 systemd[961]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
mar 01 10:52:42 rsp_tvventas1 systemd[961]: Reached target Sockets.
mar 01 10:52:42 rsp_tvventas1 systemd[961]: Reached target Timers.
mar 01 10:52:42 rsp_tvventas1 systemd[961]: Reached target Basic System.
mar 01 10:52:42 rsp_tvventas1 systemd[961]: Reached target Default.
mar 01 10:52:42 rsp_tvventas1 systemd[961]: Startup finished in 55ms.
mar 01 10:52:42 rsp_tvventas1 systemd[1]: Started User Manager for UID 1000.

edocastri, 2018-03-02:
Here's a new crash log:

mar 02 09:59:33 rsp_tvventas1 python[646]: update_playlist
mar 02 09:59:33 rsp_tvventas1 python[646]: Generating asset-list...
mar 02 09:59:33 rsp_tvventas1 python[646]: generate_asset_list deadline: 2017-12-14 17:36:00
mar 02 09:59:33 rsp_tvventas1 python[646]: get_next_asset after refresh
mar 02 09:59:33 rsp_tvventas1 python[646]: get_next_asset counter 0 returning asset 15 of 88
mar 02 09:59:33 rsp_tvventas1 python[646]: Showing asset Entretencion - Venom Trickshots II (video)
mar 02 09:59:33 rsp_tvventas1 python[646]: Asset URI /home/pi/screenly_assets/361825e6e31149398bd7fee7dffbc779
mar 02 09:59:33 rsp_tvventas1 python[646]: Displaying video /home/pi/screenly_assets/361825e6e31149398bd7fee7dffbc779 for 177
mar 02 09:59:33 rsp_tvventas1 python[646]: current url is file:///tmp/screenly_html/black_page.html
mar 02 10:02:32 rsp_tvventas1 python[646]: Last update: 2018-03-02 08:05:13.634484
mar 02 10:02:32 rsp_tvventas1 python[646]: get_next_asset
mar 02 10:02:32 rsp_tvventas1 python[646]: refresh_playlist
mar 02 10:02:32 rsp_tvventas1 python[646]: refresh: counter: (0) deadline (2017-12-14 17:36:00) timecur (2018-03-02 13:02:32.656681)
mar 02 10:02:32 rsp_tvventas1 python[646]: update_playlist
mar 02 10:02:32 rsp_tvventas1 python[646]: Generating asset-list...
mar 02 10:02:32 rsp_tvventas1 python[646]: generate_asset_list deadline: 2017-12-14 17:36:00
mar 02 10:02:32 rsp_tvventas1 python[646]: get_next_asset after refresh
mar 02 10:02:32 rsp_tvventas1 python[646]: get_next_asset counter 0 returning asset 16 of 88
mar 02 10:02:32 rsp_tvventas1 python[646]: Showing asset Info - Bolsa Reutilizable 02 (image)
mar 02 10:02:32 rsp_tvventas1 python[646]: Asset URI /home/pi/screenly_assets/2ccb70ee9eb94d33b15482272a59bc5f
mar 02 10:02:32 rsp_tvventas1 python[646]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
mar 02 10:02:32 rsp_tvventas1 python[646]: Sleeping for 25
mar 02 10:02:57 rsp_tvventas1 python[646]: Last update: 2018-03-02 08:05:13.634484
mar 02 10:02:57 rsp_tvventas1 python[646]: get_next_asset
mar 02 10:02:57 rsp_tvventas1 python[646]: refresh_playlist
mar 02 10:02:57 rsp_tvventas1 python[646]: refresh: counter: (0) deadline (2017-12-14 17:36:00) timecur (2018-03-02 13:02:57.777745)
mar 02 10:02:57 rsp_tvventas1 python[646]: update_playlist
mar 02 10:02:57 rsp_tvventas1 python[646]: Generating asset-list...
mar 02 10:02:57 rsp_tvventas1 python[646]: generate_asset_list deadline: 2017-12-14 17:36:00
mar 02 10:02:57 rsp_tvventas1 python[646]: get_next_asset after refresh
mar 02 10:02:57 rsp_tvventas1 python[646]: get_next_asset counter 0 returning asset 17 of 88
mar 02 10:02:58 rsp_tvventas1 python[646]: Showing asset Clima en Santiago 3 (webpage)
mar 02 10:02:58 rsp_tvventas1 python[646]: Asset URI http://weather.srly.io/
mar 02 10:02:58 rsp_tvventas1 python[646]: current url is http://weather.srly.io/
mar 02 10:02:58 rsp_tvventas1 python[646]: Sleeping for 25
mar 02 10:03:23 rsp_tvventas1 python[646]: Last update: 2018-03-02 08:05:13.634484
mar 02 10:03:23 rsp_tvventas1 python[646]: get_next_asset
mar 02 10:03:23 rsp_tvventas1 python[646]: refresh_playlist
mar 02 10:03:23 rsp_tvventas1 python[646]: refresh: counter: (0) deadline (2017-12-14 17:36:00) timecur (2018-03-02 13:03:23.216098)
mar 02 10:03:23 rsp_tvventas1 python[646]: update_playlist
mar 02 10:03:23 rsp_tvventas1 python[646]: Generating asset-list...
mar 02 10:03:23 rsp_tvventas1 python[646]: generate_asset_list deadline: 2017-12-14 17:36:00
mar 02 10:03:23 rsp_tvventas1 python[646]: get_next_asset after refresh
mar 02 10:03:23 rsp_tvventas1 python[646]: get_next_asset counter 0 returning asset 18 of 88
mar 02 10:03:23 rsp_tvventas1 python[646]: Showing asset SIA Corporativo (video)
mar 02 10:03:23 rsp_tvventas1 python[646]: Asset URI /home/pi/screenly_assets/4e51398da92c4dc6a8557f426783da23
mar 02 10:03:23 rsp_tvventas1 python[646]: Displaying video /home/pi/screenly_assets/4e51398da92c4dc6a8557f426783da23 for 115
mar 02 10:03:23 rsp_tvventas1 python[646]: current url is file:///tmp/screenly_html/black_page.html
mar 02 10:04:59 rsp_tvventas1 systemd[1]: Starting Screenly Network Watchdog...
mar 02 10:05:01 rsp_tvventas1 python[2580]: Starting net_watchdog.
mar 02 10:05:01 rsp_tvventas1 python[2580]: Found wifi interface wlan0
mar 02 10:05:01 rsp_tvventas1 python[2580]: Starting new HTTP connection (1): www.screenlyapp.com
mar 02 10:05:01 rsp_tvventas1 python[2580]: Traceback (most recent call last):
mar 02 10:05:01 rsp_tvventas1 python[2580]:   File "/usr/sbin/screenly_net_watchdog.py", line 131, in <module>
mar 02 10:05:01 rsp_tvventas1 python[2580]:     reaches_internet = http_test('http://www.screenlyapp.com')
mar 02 10:05:01 rsp_tvventas1 python[2580]:   File "/usr/sbin/screenly_net_watchdog.py", line 36, in http_test
mar 02 10:05:01 rsp_tvventas1 python[2580]:     r = requests.head(host, allow_redirects=True)
mar 02 10:05:01 rsp_tvventas1 python[2580]:   File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 97, in head
mar 02 10:05:01 rsp_tvventas1 python[2580]:     return request('head', url, **kwargs)
mar 02 10:05:01 rsp_tvventas1 python[2580]:   File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 57, in request
mar 02 10:05:01 rsp_tvventas1 python[2580]:     return session.request(method=method, url=url, **kwargs)
[... cut ...]
mar 02 10:25:37 rsp_tvventas1 systemd[2641]: Listening on GnuPG network certificate management daemon.
mar 02 10:25:37 rsp_tvventas1 systemd[2641]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
mar 02 10:25:37 rsp_tvventas1 systemd[2641]: Reached target Sockets.
mar 02 10:25:37 rsp_tvventas1 systemd[2641]: Reached target Basic System.
mar 02 10:25:37 rsp_tvventas1 systemd[2641]: Reached target Default.
mar 02 10:25:37 rsp_tvventas1 systemd[2641]: Startup finished in 92ms.
mar 02 10:25:37 rsp_tvventas1 systemd[1]: Started User Manager for UID 1000.

vpetersson, 2018-03-02:
ping @antonmolodykh

rusko124, 2018-09-27:
Hi @edocastri ! Can u tell me, what version of screenly you use? And could you try use latest version??