Screenly/Anthias #757: "Unable to resolve local IP address" error on startup when using a different access point with the same SSID

mikebenner, 2018-03-02:
On initial config I choose an SSID and enter authentication information. I can cycle power and join the same SSID with the access point. When I reboot the unit and move out of range of the initially configured access point but still within range of another access point with the same SSID I get the error "Unable to resolve local IP address"

I believe that the application should still join the wireless network with a different AP because it has the same SSID and authentication information.

Using latest image from NOOBS as of 3-2-18 on Raspberry Pi 3 Model B. I use a Cisco Wireless Controller with multiple access points that share the same SSID.

antonmolodykh, 2018-03-15:
I'm not sure that this would be the right way. But anyway, we use Resin Wifi Connect and are limited by its means, so at the moment we cannot do it.
At the moment to select another wifi network you need to run `rm ~/.screenly/wifi_set` and then reboot a device (without a connection). 
We are working to make it more convenient

tentact, 2018-03-16:
I have Raspian and Screenly on my RP the raspian OS is connected to my WiFi but when I go to Screenly I get unable to resolve local IP address does Screenly have a WiFi setting to connect it to WiFi I assumed if you did it through raspian you wouldn’t have to redo it in Screenly

vpetersson, 2018-03-16:
@tentact I'm not sure I understand. Screenly is designed to run on e dedicated device, so I'm not sure what you mean by that it works in Rasbpian. You can configure networking multiple ways. See [this](https://github.com/Screenly/screenly-ose/issues/746) thread for more details.

tentact, 2018-03-16:
I meant the WiFi works on raspian but when I switch over to Screenly it doesn’t

vpetersson, 2018-03-16:
Screenly OSE is based on Raspbian, so that's strange. Moreover, you can always configure WiFi manually on Raspberry Lite and then install Screenly OSE by hand if need.

antonmolodykh, 2018-03-20:
Guys, we fixed it. Now if there is no connection, the OSE will prompt you to select a wifi network.
If you don't need it (for example you want to run OSE in offline mode), then you can enable offline mode (through a settings in the web interface).
Please run an update (development version).

tentact, 2018-03-22:
mine app still says unable to resolve ip

antonmolodykh, 2018-03-22:
@tentact  it's strange, now if there is no connection, then in any case resin-wifi-connect should start (if you didn't make `enable_offline_mode=True` in the settings). are you sure that the update went ok?

tentact, 2018-03-22:
Haven’t touched the WiFi settings I don’t even know how to access it is it through the prompt? Also how do you run the update? I was just going to reinstall Screenly but it looks like the newest version of Screenly on here says January

antonmolodykh, 2018-03-22:
just run `bash <(curl -sL https://www.screenly.io/install-ose.sh)` and select the development version

tentact, 2018-03-26:
run it where from the raspberry pi?  

i dont know if its me or not but this seems way more harder than it needs to be seems like it should simply be a setting in screenly to chose/config wifi.   like raspian

vpetersson, 2018-03-26:
> run it where from the raspberry pi?

Yes.

Contributions are welcome to simplify the process, but we do have on our roadmap so simplify the upgrade process, but it’s blocked by a few other things right now.

tentact, 2018-03-26:
Okay how do you run the bash update

antonmolodykh, 2018-03-27:
@tentact  you need to run `bash <(curl -sL https://www.screenly.io/install-ose.sh)` on your Pi. 

You can connect to a Pi using SSH (SSH is disabled by default on Raspbian, you can find instruction on how to enable it [here](https://www.raspberrypi.org/documentation/remote-access/ssh/)). 
Or you can connect the keyboard to the device, go to the terminal (press ctrl+alt+F1) and enter the command there.

mikebenner, 2018-03-27:
Confirmed and tested with the development branch, this is a good fix for my issue. One tiny item though, the error message currently says "No network connection were found". I think this should be changed to "No network connection was found".

tentact, 2018-03-29:
Got the bash update and everything completed.  I get the “no network we’re found a access point was created “
So I connect to the access point put the in the password then connect my WiFi after that the pop  says “applying changes your device will soon be online....”  but the pop up doesn’t change and the device never goes online

antonmolodykh, 2018-04-02:
@tentact If the access point is still active, it may mean that you entered the password incorrectly. can you check again and confirm that the problem is not this?

tejasbingo, 2018-05-04:
I apologize in advance for the good question.  I get the same screen as tentact.  When I connect through my laptop, there is no dialog box that pops up.  I'm stuck...  Let me know what other info tho provide.

tentact, 2018-05-04:
I figured out you can only connect to one WiFi if you try to take it somewhere else after you do you will get the unable to resolve IP screen. So after you install don’t set up the WiFi until you are where you want it to stay

vpetersson, 2018-05-04:
@tentact - We've introduced a new feature that allows you to reset this configuration. You will however need to do this before leaving the network while you still have access to the node, or connect a keyboard and delete the file `.screenly/initialized`.

closed 2018-05-04 by vpetersson