Screenly/Anthias #758: Request: Disable "Check for Update"

amo28, 2018-03-03:
Hello,
I'd like to find out if it'd be possible to add a value to screenly.conf to disable checking for updates.
I have a few units deployed on networks where github and pypa.io are blocked. 

I'm running Spring 6 on them and getting the same experience as [Issue 740](https://github.com/Screenly/screenly-ose/issues/740). During the daily check for updates, the viewer goes into a bad state. I'm trying to update to the dev version remotely, but without github access, it's challenging to run the install and upgrade scripts in general.

vpetersson, 2018-03-05:
@amo28 I think the issue that you're running into has been fixed in the development version. Could you upgrade to the development version (re-run the upgrade and select yes to the development version) to see if the issue persists?

amo28, 2018-03-05:
Thanks Viktor. I've tested the dev version in a non-networked state and it definitely works if it's fully offline. 

My issue is a little difficult in that DNS allows git commands to resolve to some extent, but doesn't allow the traffic or transaction to complete. Any attempt to reach the repo comes back with a "handshake failed" after a long timeout. So I'm not even able to run the upgrade script.

What I did do to work around this was copy a zip of the repo directly to the pi, extract it to the /home/screenly folder and re-ran the install.sh, with lines commented out to pull the repo and the curl script going to pypa.io. I also commented the line that populates latest_screenly_sha. This allows the install to complete.

I then commented remaining git commands found in viewer.py, server.py, and a few others. Basically commenting any portion that attempted to reach the repo, but leaving in "else" code to allow it to fail the update and move to the next step. I would just start screenly, find the next error in journalctl, and then comment out the next git command it was trying to run. I'm at the point now where it is working fine after two days.

I don't know if this request would benefit others, but I'm really just asking if it's possible to make this a configurable option. Allow the Pi to rely on github, or a locally stored repo instead. I am deploying these in locations where the network proprietors are increasingly critical of the type of network traffic they allow.

vpetersson, 2018-03-05:
Yeah that does sound like an edge case tbh. If you're operating in a sensitive environment, I'd either make them offline entirely and block all outbound traffic, or put the devices on a dedicated VLAN with no access to anything other than the outside world.

closed 2018-03-13 by vpetersson

cpence, 2018-04-08:
I know this is a closed issue, but I'm getting exactly the same thing at a university-based installation. "Building devices," like digital signage, are connected to a VLAN that has internal access but no internet. I really don't have an option to change that status, so I'm going to have to make the same manual hacks that @amo28 mentioned.

A setting to disable all outbound internet traffic in Screenly would really be appreciated.

antonmolodykh, 2018-04-09:
@crdotson @amo28  okay guys, I understand with what moments you are facing, but it's hard for me to be sure, without having the opportunity to look at it. I want to look at the logs (before your code changes). can you attach it? just run `sudo journalctl > foobar.txt`

cpence, 2018-04-09:
I thought that I saw that offline mode disables the web server. I don't want that, either -- I still have internal network access to the machine, so I want to be able to use the web interface.

And I've got NTP set up. (Notably, though this is a matter for another bug report, NTP isn't installed by default on the latest Screenly OSE images, which it probably should be.)

But! After scrolling through the code, it is perfectly possible to fix this behavior as the code sits now. There's an environment variable, `DISABLE_UPDATE_CHECK`, which works perfectly. You need to:

1. Edit `screenly-viewer.service`, add `Environment=DISABLE_UPDATE_CHECK=1`.
2. Delete `~/.screenly/latest_screenly_sha`. If this file is missing, the server will just silently abandon update checking.
3. (Make sure you're doing something for clock sync, or your schedules will all fail.)

And everything works!

amo28, 2018-04-09:
@cpence Wish I had known of that variable in screenly-viewer.service before. That sounds exactly like what I was looking for. And the offline mode did not affect the web server for me.

@antonmolodykh At the location where I was having the issue, I don't have any units that do not have modified code unfortunately. But for me it boiled down to the unit being able to resolve github, but due to network restrictions any git commands would fail or time out, and the update process did not appear to fail gracefully. I know log would be more helpful but I didn't save them. Sorry about that.

vpetersson, 2018-04-10:
@cpence 

> Notably, though this is a matter for another bug report, NTP isn't installed by default on the latest Screenly OSE images, which it probably should be

We should double check this, but I'd be very surprised if it isn't. Since the Raspberry Pi lacks a Real Time Clock (RTC), if it is unable to connect to an NTP server, the date and time would be off by a lot.