Screenly/Anthias #760: Resin WiFi not accepting password

nwoflord, 2018-03-12:
Using the new image I have installed it and the Pi boots correctly and brings up the Resin screen.  It broadcasts the network but when I attempt to connect to it, it keeps telling me that the password is invalid. I have tried with different devices with the same issue.  Has anyone else had this issue or is it something that I'm doing wrong.

nwoflord, 2018-03-12:
I did check that because I have had a similar issue before.  That however was not the case.  I was able to get one to take and let me connect to the network after several reboots, hoping that the second one goes better.  Thank you for you help!

vpetersson, 2018-03-13:
Very strange. The only thing I can think of is if it is a range issue. The antenna on the Raspberry Pi isn't that strong.

closed 2018-03-13 by vpetersson

nreppa, 2018-06-07:
I'm having this same issue with multiple broadcasted networks. Issue isn't range as the pi is sitting on my desk. Doesn't work to connect from computer or mobile device.

ACEDornback, 2019-12-18:
Same issue.
Latest OSE img.
Tried multiple devices.
Resetting pi each time.
Also tried creating fresh sd card each time.
Tried different wifi USB.
I can see the access point.
But incorrect password and then unable to connect after first wifi connect attempt and the SSID disappears.

Is there a default developer password that was used in creating/building the original image that we can try?

iCodeOkay, 2020-01-08:
Ditto

iCodeOkay, 2020-01-31:
Just wanted to add that I had this issue with a Raspberry Pi Zero-W. Have the same issue with a Raspberry Pi 3

vpetersson, 2020-02-05:
@iCodeOkay Could you try upgrading to the latest development version to see if that solves the issue? We've upgraded the component that deals with this.