Screenly/Anthias #764: Enable offline mode

antonmolodykh, 2018-03-15:
We need to add a new item in the settings, something like "enable offline mode". 
By default if there is no connection, it will be offered to select wifi network (using resin wifi), but if this option is enabled, assets will just start to be displayed.

ParrainNN, 2019-04-05:
Hi @vpetersson !

I will interested by this feature because I would like to use Screenly OSE as a complete standalone solution (no Internet connected) and managed only by mobile devices.  So this feature should propose to enable and configure an access-point (hidden or not / SSID / passphrase / security type / IP range).

Thanks!

vpetersson, 2019-04-05:
@ParrainNN It's not that simple. You can't have the Pi operate like that without adding additional hardware (namely an RTC). Because of this, we have not pursued this path. In "offline" mode as instructed here, you'd still net a network connection to connect to an NTP server to set the time.

mlott, 2019-04-27:
@ParrainNN The offline feature you describe is exactly what I'm after.  RTC hardware is cheap and easy to attach and the current time could be updated when a device (ie Mobile) connects to the AP.    I've been updating Screenly OSE by using the Wifi Hotspot function on my mobile then connecting with a second device to upload new content.  I would love to work my own fork, but unfortunately my grasp of python far to limited.  Maybe one day!

urb5tar, 2019-05-13:
Some times you don't need the exact time. In my case i want to play a video on a exhibition stand. it is just a loop of videos and pictures. the exact time is not important. 

in my opinion there is no need for an offline option. Just display the asset list after 3 minutes.

vpetersson, 2019-05-13:
In theory, you can do this by setting the start date from 1979 and onwards (i.e. beginning of UNIX time).

urb5tar, 2019-05-13:
> 
> 
> In theory, you can do this by setting the start date from 1979 and onwards (i.e. beginning of UNIX time).

yes, but the assets wont start, because it is waiting for a network connection.

vpetersson, 2019-05-13:
We need to understand better what is causing this and if it is in the system or in the Screenly stack. If the former, there might be just some configuration option.

urb5tar, 2019-05-13:
OK I try to explain it better.

When Screenly starts, it want to have a network connection. if there is no network connection it starts a resin.io AP. Then waits endless till a connection is established. No assets are played till a connection is present.

The demanded "offline mode" is nothing but to start playing the assets whether there is a connection to a wifi or not. In a previous version this works without any further configuration.

albanoalfredo, 2019-06-29:
I don't understand why it needs a connection to update the playlist when you add assets.

Ideally, if no connection is found it should have two options:

- local areal network (without internet access)

- offline mode (reads the local assets or check for usb drives, mount it and play all the compatible files)

In both modes it would be great having a way to setup the time manually, eliminating the net for a network connection.

I'd love to help somehow but my python skills are nearly non-existent

kyureus, 2020-04-10:
I'm currently trying to do the same thing with Screenly OSE for work. I'm trying to make a scrolling task list on a TV mounted above my desk so that my team can keep track of tasks that need to be completed with out prompting me or my manager. We're in a semi-secure space so I can't get an outside internet connection, and I can't connect it to a corporate connection. I currently have a wireless router connected on my desk and I can connect from my mobile to the player through the wireless router and update files, as well as updating the files with a thumb drive, but the files won't play either way without an NTP server.

vpetersson, 2020-04-10:
> I'm currently trying to do the same thing with Screenly OSE for work. I'm trying to make a scrolling task list on a TV mounted above my desk so that my team can keep track of tasks that need to be [...]

You'll need either an NTP server *or* an RTC device HAT for the Raspberry Pi. There's no other workaround that I'm aware of.

kc9rge, 2020-04-17:
if there is no need for the time then there should be a workaround. I work in a secure environment. no wifi, no Bluetooth. if you want government/military contracts you will need to be able to operate in secure environments, meaning no internet.

vpetersson, 2020-04-17:
> if there is no need for the time then there should be a workaround

Well....it's digital signage. Scheduling of assets is a very core feature in this category of products. To make an analogy, try running `cron` or `systemd-timer` without a functioning clock.

urb5tar, 2020-04-17:
> > if there is no need for the time then there should be a workaround
> 
> Well....it's digital signage. Scheduling of assets is a very core feature in this category of products.

There are applications where time does not matter.  In my case, I want to display a video or pictures in a certain order at a fair.  In a continuous loop.  I only need a network connection once to upload the data.  Unfortunately there is not always internet or wifi.  It is enough to wait two minutes after the request for a network connection and then simply display the assets.

ParrainNN, 2020-04-17:
> There are applications where time does not matter. In my case, I want to display a video or pictures in a certain order at a fair. In a continuous loop. I only need a network connection once to [...]

Hi urb5tar!

In Screenly, you can set a starting Date/Time in the past for your assets. As the Raspberry will start without Internet Connection for setting its time and that it's not RTC (Real-Time Clock module) equipped, its system date will be 01/01/1970. You can certainly set a starting Date for your video or pictures in 1902, 1959 until 1969... These assets will be automatically displayed after boot.

ParrainNN, 2020-04-17:
@vpetersson , i'm sorry but you are still focusing on the "time" aspect of our request.  Time (scheduling) isn't the problem. Some of us use RTC, some others don't (and use a workaround by setting their assets with a old date/time ie 1970 and before).

We just want to use Screenly without Internet Connection and to have the ability to configure the Raspberry as a permanent Wi-Fi Hot-Spot, with DHCP server, so that we can use the Screenly GUI.

Respectfully.
ParrainNN.

vpetersson, 2020-04-17:
> @vpetersson , i'm sorry but you are still focusing on the scheduling aspect of Screenly with our request. Scheduling isn't the problem. Some of us use RTC, some others don't (and use a workaround by setting their assets with a old starting date/time ie 1970 and before).
> 
> We would just like to use Screenly without Internet connection and to have the ability to configure the Raspberry as a permanent "stand alone/local" Wi-Fi Hot-Spot, with DHCP server, so that we can use the Screenly GUI from mobile devices for instance.
> 
> Respectfully.
> ParrainNN.

That's a lot of work (in particular for an edge case). That said, we're happy to receive a pull request that addresses this.