Screenly/Anthias #765: Trigger asset change via API

xstable, 2018-03-16:
Is it possible to trigger an asset-change by API?

Why need: Let's say I have an external Webpage, which need x Time to finish an animation (but I didn't know in advanced how long it would be). So the Webpage should trigger the API to do a asset-change if finished the animation.

How could this work?

vpetersson, 2018-03-16:
@xstable yes, there's an API call for switching to the next asset. You can find the docs [here](http://ose.demo.screenlyapp.com/api/docs/). The API call would be `/api/v1/assets/control/next`.

closed 2018-03-16 by vpetersson

xstable, 2018-03-16:
Thank you, but I've looked in this API-documentation to found anything like that... but didn't.
So there is no /control/next on this page... or did I miss it?

vpetersson, 2018-03-16:
Yes, it's under "/api/v1/assets/control/{command}"

xstable, 2018-03-16:
Ok, I've searched for "next" so I did't found it in the gui. 
So thank you, this helps me a lot.