Screenly/Anthias #766: Manage multiple pi's

JonnyTech, 2018-03-17:
What is the procedure for managing multiple screenly-ose pi's? Am I missing something? Can't find any kid of central control. My setup is three pi's in a room with no internet access.

vpetersson, 2018-03-17:
There isn't any. That's what Screenly Pro does (among other things). In theory, you could script something together using the [API](http://ose.demo.screenlyapp.com/api/docs/).

closed 2018-03-17 by vpetersson

JonnyTech, 2018-03-17:
I thought that Pro needs an Internet connection. My rig is in a remote barn with no web access at all. Just need to easily manage their schedules from a central computer. Does anything exist as a starting point?

vpetersson, 2018-03-17:
> I thought that Pro needs an Internet connection

Yes, that is correct. 

> Does anything exist as a starting point?

I've seen a few projects over the years who have aspired to do something similar. I *think* [this](https://github.com/MCLD/MCLD-Screenly-OSE-Central-Control) project is the one that got furthest. That said, I haven't used it, so I cannot comment on if it works or not.

JonnyTech, 2018-03-17:
Thank you very much.