Screenly/Anthias #769: To manage this screen .......

rhodesmi, 2018-03-21:
Hi

My Screenly loads up and stays on the 'to manage the content on this screen vist http://......'

It does work fine once in a while.

Its a brand new 8Gb SD HC card (MAPLIN) 

I think this is the issue and it might not be quick enough for the application to load the MP4 it is supposed to play. Is there are a timeout in the app that could prevent it from loading the first Video?

antonmolodykh, 2018-03-21:
Hi @rhodesmi 
No, the app doesn't have a timeout. Where is this video located? Are there any assets other than this video?

rhodesmi, 2018-03-21:
The MP4 is located in the default area (uploaded via the WEBUI) its a stand alone Raspberry PI with no network connectivity. It has been working fine for 3 or 4 years, we have started experiencing this issue since the last upgrade earlier this year. It was recommended (via this group) to purchase a new SD card.

antonmolodykh, 2018-03-21:
okay, can you do `sudo journalctl > foobar.txt` and attach foobar.txt once this happens?

rhodesmi, 2018-03-21:
here you go - thanks
[foobar.txt](https://github.com/Screenly/screenly-ose/files/1833014/foobar.txt)

antonmolodykh, 2018-03-21:
Okay, it seems we have already solved this problem. Please run an update (development version) and let me know if there are any problems.

rhodesmi, 2018-03-21:
great that sound promising i will connect it to the internet and run the update command.

rhodesmi, 2018-03-21:
interestingly (or not) as soon as i plugged in the network cable she sprung into life. You mentioned (development version) how do i upgrade/patch to that version?

antonmolodykh, 2018-03-22:
just run bash <(curl -sL https://www.screenly.io/install-ose.sh). You will be asked to select the development version

vpetersson, 2018-05-04:
Closing out ticket due to inactivity.

closed 2018-05-04 by vpetersson

rhodesmi, 2018-05-04:
Hello

Sorry forgot to say thanks for all your help. It's all working really well
now.