Screenly/Anthias #770: Screenly does not start presentation cycle

Miq1, 2018-03-21:
I got Screenly OSE freshly installed from the latest image. I am able to add assets etc., but the presentation cycle will never start. I am stuck on either the splash screen or on the assets management screen, on whichever I left the system.
My current test assets are three JPG images 1080x1920 (I am using a portrait mode screeen, display_rotate=3).
What am I missing here?

antonmolodykh, 2018-03-21:
Hi, @Miq1 
I think the problem is not related to your screen mode.
I'm not sure I understand you. Web interface is available, you can add assets, but your display shows `splash page` (to manage the content on this screen visit http://yout_ip), right?

Miq1, 2018-03-21:
Exactly, that is correct. If I leave the splash page on after a reboot, it will stay. If I will click on the IP on the splash page to get to the asset management page, that will stay on.

Miq1, 2018-03-22:
Can it be that Screenly is trying to contact a server at startup to look for updates etc.? And hangs there forever, since the Raspberry is not connected to any network in our case?

antonmolodykh, 2018-03-22:
just run `bash <(curl -sL https://www.screenly.io/install-ose.sh)`. You will be asked to select the development version

antonmolodykh, 2018-03-22:
it's difficult for me to say what the problem is without logs. can you do `sudo journalctl > foobar.txt` and attach foobar.txt?

Miq1, 2018-03-22:
[foobar.txt](https://github.com/Screenly/screenly-ose/files/1837762/foobar.txt)
Here you go! :-)

antonmolodykh, 2018-03-22:
It seems that you took the logs too early. please run screenly, wait for splash screen and a couple more minutes

Miq1, 2018-03-22:
Okay, will do, but only tomorrow. Thank you for now!

Miq1, 2018-03-23:
I made some progress now. I took the Raspi home to do apt-get upgrade etc. and connected it via eth0 to my home network. After the upgrade and reboot Screenly started up, giving me the local network IP on the splash screen and voilà: started presenting.

Then I set up the wlan0 interface to work as access point (based on [https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md](url)). This is necessary because in the final environment the Raspi has no network connection, administration shall be done by connecting to it via its own Wifi.

Now the Wifi network is established at boot and I am able to connect to it, but Screenly on the splash screen now is stating "Unable to determine local IP". Thus I am unable to manage the assets now.

Where will I have to tell Screenly to use the set-up Wifi?

vpetersson, 2018-03-23:
See #755. We are working on improving that part.

Miq1, 2018-03-23:
Hmmm... Is there a manual way to fix this for the time being? Like, executing some script on cron some seconds after boot or so?

I need to have the thing run next week, so I am desparately looking for solutions.

I would provide the assets on a USB stick, but still would need access to the management page.

vpetersson, 2018-03-23:
@Miq1 That's not really a supported use case. The device would need some kind of networking (even if it cannot talk out). The offline mode referenced before is mostly for temporary outages (or just in playback mode). You'd still configure the device using the network interface.

Frankly using USB stick for assets is not what the system is designed for. It would require a fair bit of rework. Code contributions would be welcome but it isn't something that we have on the roadmap to support now.

Miq1, 2018-03-23:
Ough! =:-o

So I will not be able to run Screenly on an isolated system at all? Then I will have to look for alternative solutions, I am afraid.

vpetersson, 2018-03-23:
You can run it without internet connection on an isolated network, but you still need some kind of network connection in order to configure the system.

Miq1, 2018-03-23:
And that network connection needs to be external, the Wifi spanned by the Raspi itself will not suffice?

vpetersson, 2018-03-23:
Yes, our logic assumes that it is an external connection. In theory, you might be able to configure the the Pi to act as a hotspot but I don't think it is a very well tested use case so it is likely that you will need to dive into the code base to patch it.

Miq1, 2018-03-23:
What a pity! The Raspi setup as a hotspot is plain vanilla. I do not quite understand the difference it makes for Screenly between the Raspi working as a client in another network and the Raspi spanning a network itself. Screenly is merely a server over IP in this aspect, isn't it?

Miq1, 2018-03-23:
Thank you for clarification. I am not into Python at all, so I will not be able to get a grip at it, I am afraid.

I will again start looking for an alternative solution. Screenly is really good work, by the way, if it was not my odd environment :-)

chucklin72, 2018-10-06:
Brand new install
2018-05-04-Screenly_OSE_4GB
Raspberry Pi 3 B+
state 0x120009 [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

Doesnt get off the splashscreen except if I use Hackernews. Screenly clock and weather do not work.

root@raspberrypi:/home/pi# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Logs:
Oct 06 03:50:53 raspberrypi python[559]: Generating asset-list...
Oct 06 03:50:53 raspberrypi python[559]: generate_asset_list deadline: 2018-05-04 15:04:00
Oct 06 03:50:53 raspberrypi python[559]: get_next_asset after refresh
Oct 06 03:50:53 raspberrypi python[559]: get_next_asset counter 0 returning asset 1 of 1
Oct 06 03:51:13 raspberrypi python[559]: /usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
Oct 06 03:51:13 raspberrypi python[559]:   InsecureRequestWarning)
Oct 06 03:51:13 raspberrypi python[559]: Showing asset Screenly Clock Widget (webpage)
Oct 06 03:51:13 raspberrypi python[559]: Asset URI https://clock.srly.io/
Oct 06 03:51:13 raspberrypi python[559]: Already showing https://clock.srly.io/, reloading it.
Oct 06 03:51:13 raspberrypi python[559]: Sleeping for 10
Oct 06 03:51:23 raspberrypi python[559]: Last update: 2018-10-06 03:04:39.568396
Oct 06 03:51:24 raspberrypi python[559]: get_next_asset
Oct 06 03:51:24 raspberrypi python[559]: refresh_playlist
Oct 06 03:51:24 raspberrypi python[559]: refresh: counter: (0) deadline (2018-05-04 15:04:00) timecur (2018-10-06 03:51:24.045136)
Oct 06 03:51:24 raspberrypi python[559]: update_playlist

screenly.conf
root@raspberrypi:/home/pi/.screenly# cat screenly.conf 
[viewer]
verify_ssl = off
default_streaming_duration = 300
player_name = test player
debug_logging = on
show_splash = on
resolution = 1920x1080
audio_output = hdmi
shuffle_playlist = off
use_24_hour_clock = off
default_duration = 10

[main]
assetdir = /home/pi/screenly_assets
database = /home/pi/.screenly/screenly.db
websocket_port = 9999
use_ssl = off
analytics_opt_out = off
use_24_hour_clock = off

[auth]
password = 
user =

chucklin72, 2018-10-06:
Shorter video ( 11 sec ) instead of 2 minute with G2 GL worked. still cannot get web to work.

vpetersson, 2018-10-08:
@chucklin72 Are you able to reproduce this on the latest development build?

ShukhratK, 2019-05-20:
Hi,

I am having the same issue here. Cant get past the splash screen that keeps restarting.