Screenly/Anthias #772: Switch to Predictable First-Run Wifi Password

mrampersad, 2018-03-23:
During my troubleshooting for #755, I did not have a TV or wired ethernet adapter and it was impossible to configure a stock Screenly image on a Pi Zero W without these. As a workaround, I wrote the image, mounted it on my desktop, and modified the SSID generation script to always generate the same hardcoded password. Then I waited for a random Screenly network to appear and connected with the known password.

If it would not be too much of a security risk, please make the resin-wifi-connect password equal to the MAC address of wlan0, or the first MAC address available in the network adapter list.

vpetersson, 2018-03-23:
@mrampersad Interesting idea. Yeah configuring the device entirely without networking can be bit challenging. I think an easier workaround would be to simply configure the WiFi using `/etc/network/interfaces` and subsequently bypass the hotspot model entirely.

mrampersad, 2018-03-24:
I failed to mention that I run Windows primarily, and mounting the image to edit the files is not easy. I also avoided making changes to non-screenly files because I wasn't sure if /etc/network/interfaces was the correct file or maybe NetworkManager or wpa_supplicant would be the right place to edit. I also do not know if the Screenly database would overwrite unexpected system-level changes. Asking for support at such a low level would be unfair.

A completely static password that is not based on any unique feature of the device would also suffice. Something like 'screenly'.

Many embedded device manufacturers have a default username and password for the first-run. With resin-wifi-connect you are forced to connect to a new network as your first action. I believe the reduction in security is acceptable, but I understand if you disagree. The SSID should remain random to allow many devices to be setup in the same vicinity at the same time.

vpetersson, 2018-08-13:
@mrampersad Perhaps a better solution is something like this:

 * Create a file called 'wifi-password' on the boot partition (which is readable from Windows) with a password.
 * The system will then look for this file during initiation. If the file exists, it will use it.
 * When done, it will delete the file for security reasons.

What do you think? I'd be happy to accept such pull request.

pmb-sd, 2018-11-28:
I had an issue where for some reason the configured wifi was lost, and it switched over to try and help me set up again. but with the Splash screen disabled, you can't see the newly generated password that goes with the temp wifi...  so, i was essentially locked out of changes. I couldn't find any answers for how to configure things manually either. (where do can someone ask simple questions??)
anyway, some other way to even view this temp password would be welcome! maybe it can be accessible from the console?
that brings up another issue, so even if I could see the temp password in console is was a PAIN trying to do anything in there, because just videos keep playing over the top of the console after CTRL + ALT + F1... I will continue to look for the answer to that (how to stop playback without access to the interface), before opening a ticket about it...

rusko124, 2018-11-29:
Hi, @pmb-sd !
I can suggest you take the following steps:
- When loading Screenly quickly try to go to the console (CTRL + ALT + F1) and enter the following command `sudo systemctl stop screenly-viewer`
- Then set `show_splash = on` in `/home/pi/.screenly/screenly.conf`
- Reboot

pmb-sd, 2018-11-29:
Thanks @rusko124, that looks like the answer I needed!
this goes in my notes.