Screenly/Anthias #773: RasPi 3 B+

JonnyTech, 2018-03-26:
Is the current image (2018-01-19-Screenly_OSE_4GB.zip) compatible with the new Raspberry Pi 3 B+? I have dd'ed the image onto a SD card but upon boot it just shows the rainbow splash screen.

[image]

zekikoese, 2018-03-26:
I have the same issue. Tried older releases too, none of them work tough. 
Now I tried to install screenly manually and everything seemed to install fine but after the mandatory reboot screenly doesn't start and the terminal is shown.

vpetersson, 2018-03-26:
This is expected. The Raspberry Pi 3 Model B+ requires a newer Raspbian version. We will provide a new build at the end of this sprint. In the meantime, if you want to run Screenly OSE on the B+, you would need to install it by hand on the latest Raspbian image.

JonnyTech, 2018-03-26:
Thanks for the reply @vpetersson, if I create a Raspbian Lite SD is there a list additions to install?

vpetersson, 2018-03-26:
@JonnyTech Yes, you can find the instructions [here](https://www.screenly.io/ose/) under Option 3.

JonnyTech, 2018-03-26:
How did I miss that?! Thanks @vpetersson. I shall report back if any issues.

JonnyTech, 2018-03-26:
All done. The only additional step required was `sudo apt install -y network-manager` after first boot.

closed 2018-03-26 by JonnyTech