Screenly/Anthias #774: Backup not working in Resin.io implementation

McAleers, 2018-03-26:
Screenly OSE installed on Raspberry Pi 3B, using Resin.io for deployment. When I go to the settings screen and click on the "Get Backup" button, I receive the following error:

`Server Error: [Errno 2] No such file or directory: '/data/screenly/static/screenly-backup.tar.gz'`

I've replicated on multiple instances - I can't get a backup image from any of them.

Thanks for any help.

vpetersson, 2018-03-26:
Interesting find @McAleers. What resinos version are you using?

McAleers, 2018-03-26:
Hi, I'm using:

HOST OS VERSION
Resin OS 2.12.3+rev1 (prod)

SUPERVISOR VERSION
7.1.14

timoa, 2018-04-17:
Hi,

Same here:

**OS**: Resin OS 2.12.0+rev1 (prod)
**Supervisor**: 7.1.7

It looks like that it tries to save the export in a read-only folder (Docker) and the file doesn't exist when 
we try to download it.

In the logs (debug mode ON):
```
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: Traceback (most recent call last):
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: File "/home/pi/screenly/server.py", line 239, in api_view
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: return view(*args, **kwargs)
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: File "/usr/local/lib/python2.7/dist-packages/flask_restful_swagger_2/swagger.py", line 219, in inner
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: return f(self, *args, **kwargs)
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: File "/home/pi/screenly/server.py", line 608, in post
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: filename = backup_helper.create_backup()
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: File "/home/pi/screenly/lib/backup_helper.py", line 23, in create_backup
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: remove(file_path)
17.04.18 07:30:13 (+0100)  main  Apr 17 06:30:13 ******* python[205]: OSError: [Errno 2] No such file or directory: '/data/screenly/static/screenly-backup.tar.gz'
```

antonmolodykh, 2018-04-17:
@McAleers @timoa  we know that this is an actual problem. thanks for the reports guys. I'll let you know when we fix it.

vpetersson, 2018-05-04:
Duplicate of #800.

closed 2018-05-04 by vpetersson