Screenly/Anthias #780: Google Data Studio Dashboard Not loading completely

Andrew-619, 2018-03-27:
I am trying to display a Google data studio dashboard with 5 charts/graphs. Only 3 of the 5 graphs load and the rest doesn't. However, when i look at this web page from my computer it shows all the graphs perfectly. 

I tried making multiple dashboards and this happens in all the cases. 

Does the browser have some sort of limitation ? What could i do to fix this

antonmolodykh, 2018-03-28:
Probably this is due to uzbl, which we are now using. We are working on replacing the current browser.

closed 2018-05-04 by vpetersson

antonmolodykh, 2018-06-20:
@Andrew-619  we published the experimental screenly ose version. one of the major changes is the new browser. To try it follow the steps under 'Manual upgrade' here and answer yes to the question about experimental version.
We are looking forward to feedback, please let us know about any problems.