Screenly/Anthias #783: Display random asset

JonnyTech, 2018-03-28:
The little old ladies at the local art group are loving the system! They are using it in the village church for their art exhibition and thoroughly enjoying setting it up and seeing the results. It is strange seeing them use a computer. That shows what a great system you have created.

Anyway, they have yet another request: Can a random image be displayed from the assets? They have imported hundreds of images and would like to show a few at random without having to create multiple pseudo random lists.

antonmolodykh, 2018-03-29:
Hi @JonnyTech glad to hear that!
Screenly OSE has the ability to display assets in a random order. For this go to the settings (via the web interface) and choose "Shuffle playlist"

JonnyTech, 2018-03-29:
Excellent - they must have missed that - thanks!

But we have another problem, I shall raise another issue...

closed 2018-03-29 by JonnyTech