Screenly/Anthias #784: Pi locked up overnight

JonnyTech, 2018-03-29:
Screenly OSE on a Pi 3 B+ was running a playlist cycling 100 images yesterday. It was left running overnight (someone forgot to turn it off before leaving). This morning when turning on the screen it was stuck on one of the images. The Pi is still on but is not responding to pings, so cannot get to the network interface or SSH to it. Tried to switch to another TTY using CTRL+ALT+1,2,3 etc, with a local keyboard but no response. On the Pi, the LEDs on are PWR (red, solid), ACT (green, solid), and the network activity LED (yellow, flashing). I have left it on in case of any suggestions.

JonnyTech, 2018-03-29:
I have now powered the unit down and removed the sdcard. Here are the last few entries from logs:

```
$ ls -la
total 4068
drwxr-xr-x  5 root root    4096 Mar 28 07:25 .
drwxr-xr-x 12 root root    4096 Mar 26 17:40 ..
-rw-r--r--  1 root root    4137 Mar 26 17:26 alternatives.log
drwxr-xr-x  2 root root    4096 Mar 26 17:39 apt
-rw-r-----  1 root adm    91163 Mar 28 08:17 auth.log
-rw-r--r--  1 root root    1002 Mar 26 17:41 boot.log
-rw-r--r--  1 root root       0 Mar 13 22:53 bootstrap.log
-rw-------  1 root utmp     384 Mar 27 17:08 btmp
-rw-r-----  1 root adm  1339392 Mar 28 09:06 daemon.log
-rw-r-----  1 root adm     3741 Mar 26 17:41 debug
-rw-r--r--  1 root root  188499 Mar 26 17:40 dpkg.log
-rw-r--r--  1 root root    2664 Mar 26 17:26 faillog
-rw-r--r--  1 root root    1123 Mar 26 17:26 fontconfig.log
-rw-r-----  1 root adm   162862 Mar 28 09:01 kern.log
-rw-rw-r--  1 root utmp  292292 Mar 27 17:08 lastlog
-rw-r-----  1 root adm   355830 Mar 28 09:01 messages
drwxr-xr-x  2 root adm     4096 Mar 28 07:25 nginx
drwxr-x---  2 root adm     4096 Nov 20 22:24 samba
-rw-r-----  1 root adm   135168 Mar 28 09:06 syslog
-rw-r-----  1 root adm  1570932 Mar 28 07:25 syslog.1
-rw-r-----  1 root adm   197744 Mar 28 09:01 user.log
-rw-rw-r--  1 root utmp    8448 Mar 27 17:09 wtmp
-rw-r--r--  1 root root   12631 Mar 26 17:41 Xorg.0.log

$  tail -30 daemon.log 
Mar 28 08:04:29 raspberrypi wpa_supplicant[428]: wlan0: Failed to initiate sched scan
Mar 28 08:04:33 raspberrypi python[569]: Generating asset-list...
Mar 28 08:04:33 raspberrypi python[569]: Showing asset 04.png (image)
Mar 28 08:04:33 raspberrypi python[569]: Sleeping for 10
Mar 28 08:04:43 raspberrypi python[569]: Generating asset-list...
Mar 28 08:04:43 raspberrypi python[569]: Showing asset 05.png (image)
Mar 28 08:04:43 raspberrypi python[569]: Sleeping for 10
Mar 28 08:04:53 raspberrypi python[569]: Generating asset-list...
Mar 28 08:04:53 raspberrypi python[569]: Showing asset 06.png (image)
Mar 28 08:04:53 raspberrypi python[569]: Sleeping for 10
Mar 28 08:05:03 raspberrypi python[569]: Generating asset-list...
Mar 28 08:05:03 raspberrypi python[569]: Showing asset 07.png (image)
Mar 28 08:05:03 raspberrypi python[569]: Sleeping for 10
Mar 28 08:05:14 raspberrypi python[569]: Generating asset-list...
Mar 28 08:05:14 raspberrypi python[569]: Showing asset 08.png (image)
Mar 28 08:05:14 raspberrypi python[569]: Sleeping for 10
Mar 28 08:05:24 raspberrypi python[569]: Generating asset-list...
Mar 28 08:05:24 raspberrypi python[569]: Showing asset 09.png (image)
Mar 28 08:05:24 raspberrypi python[569]: Sleeping for 10
Mar 28 08:05:32 raspberrypi wpa_supplicant[428]: wlan0: Failed to initiate sched scan
Mar 28 08:05:34 raspberrypi python[569]: Generating asset-list...
Mar 28 08:05:34 raspberrypi python[569]: Showing asset 10.png (image)
Mar 28 08:05:34 raspberrypi python[569]: Sleeping for 10
Mar 28 08:05:44 raspberrypi python[569]: Generating asset-list...
Mar 28 08:05:44 raspberrypi python[569]: Showing asset 11.png (image)
Mar 28 08:05:44 raspberrypi python[569]: Sleeping for 10
Mar 28 08:05:54 raspberrypi python[569]: Generating asset-list...
Mar 28 08:05:54 raspberrypi python[569]: Showing asset 12.png (image)
Mar 28 08:05:54 raspberrypi python[569]: Sleeping for 10

$ tail -30 syslog
Mar 28 08:04:29 raspberrypi wpa_supplicant[428]: wlan0: Failed to initiate sched scan
Mar 28 08:04:33 raspberrypi python[569]: Generating asset-list...
Mar 28 08:04:33 raspberrypi python[569]: Showing asset 04.png (image)
Mar 28 08:04:33 raspberrypi python[569]: Sleeping for 10
Mar 28 08:04:43 raspberrypi python[569]: Generating asset-list...
Mar 28 08:04:43 raspberrypi python[569]: Showing asset 05.png (image)
Mar 28 08:04:43 raspberrypi python[569]: Sleeping for 10
Mar 28 08:04:53 raspberrypi python[569]: Generating asset-list...
Mar 28 08:04:53 raspberrypi python[569]: Showing asset 06.png (image)
[... 55 lines cut ...]
Mar 28 08:01:17 raspberrypi kernel: [141613.929586] brcmfmac: power management disabled
```

The only other file on the card created at about time of last activity was a zero byte `screenly.watchdog` file in the `/tmp` folder:

```
$ ls -la
total 48
drwxrwxrwt  9 root root 4096 Mar 28 08:17 .
drwxr-xr-x 21 root root 4096 Mar 26 17:03 ..
drwxrwxrwt  2 root root 4096 Mar 26 17:41 .font-unix
drwxrwxrwt  2 root root 4096 Mar 26 17:41 .ICE-unix
-rw-r--r--  1 user user   73 Mar 26 17:57 omxplayerdbus.pi
-rw-r--r--  1 user user    5 Mar 26 17:57 omxplayerdbus.pi.pid
drwxr-xr-x  2 user user 4096 Mar 26 17:41 screenly_html
-rw-r--r--  1 user user    0 Mar 28 09:06 screenly.watchdog
drwx------  3 root root 4096 Mar 26 17:41 systemd-private-adbcf7a025d945929425bc8129a5a986-systemd-timesyncd.service-7ggNXc
drwxrwxrwt  2 root root 4096 Mar 26 17:41 .Test-unix
-r--r--r--  1 root root   11 Mar 26 17:41 .X0-lock
drwxrwxrwt  2 root root 4096 Mar 26 17:41 .X11-unix
drwxrwxrwt  2 root root 4096 Mar 26 17:41 .XIM-unix

```

I cannot see anything obvious. Any ideas?

vpetersson, 2018-04-03:
@JonnyTech Nothing obvious stands out unfortunately. In theory, `screenly.watchdog` should [only be written](https://github.com/Screenly/screenly-ose/blob/master/viewer.py#L299) when the viewer loop works.

Could you adjust the playback (fewer/different assets) to see if that makes a difference?

JonnyTech, 2018-04-04:
It seems intermittent, some units last for days, others do not. Asset amounts and types seem irrelevant. For example one unit ran for 5 days then after a power cycle it crashed after 20 hours; another unit stopped after 6 hours but after a reboot is still going after 5 days. I am manually cycling power every morning and keeping my fingers crossed!

JonnyTech, 2018-04-04:
Could it be related to an update check? The units are not connected to the Internet. I do not know at what time it checks and if I can be disabled.

vpetersson, 2018-04-04:
Interesting. Are you running on the development release or on the production release?

JonnyTech, 2018-04-04:
IIRC the development branch was recommended as I am using Pi3 B+ units.

JonnyTech, 2018-05-10:
This morning I had a black screen but was still able to ping and ssh to it. The web interface displayed:
```
502 Bad Gateway
nginx
```
Source:
```
<html>
<head><title>502 Bad Gateway</title></head>
<body bgcolor="white">
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx</center>
</body>
</html>
```

Looking at the logs:

```
pi@raspberrypi:~ $ cd /var/log

pi@raspberrypi:/var/log $ tail -30 daemon.log 
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 272, in browser_url
May 10 07:53:34 raspberrypi python[1240]:     browser_send('uri ' + escaped_url, cb=cb)
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 252, in browser_send
May 10 07:53:34 raspberrypi python[1240]:     load_browser()
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 238, in load_browser
May 10 07:53:34 raspberrypi python[1240]:     browser_send(uzbl_rc)
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 248, in browser_send
May 10 07:53:34 raspberrypi python[1240]:     if cb(browser.next()):
May 10 07:53:34 raspberrypi python[1240]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 356, in next
May 10 07:53:34 raspberrypi python[1240]:     raise StopIteration()
May 10 07:53:34 raspberrypi python[1240]: StopIteration
May 10 07:53:34 raspberrypi python[1240]: Traceback (most recent call last):
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 458, in <module>
May 10 07:53:34 raspberrypi python[1240]:     main()
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 439, in main
May 10 07:53:34 raspberrypi python[1240]:     browser_url(url=url)
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 272, in browser_url
May 10 07:53:34 raspberrypi python[1240]:     browser_send('uri ' + escaped_url, cb=cb)
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 252, in browser_send
May 10 07:53:34 raspberrypi python[1240]:     load_browser()
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 238, in load_browser
May 10 07:53:34 raspberrypi python[1240]:     browser_send(uzbl_rc)
May 10 07:53:34 raspberrypi python[1240]:   File "/home/pi/screenly/viewer.py", line 248, in browser_send
May 10 07:53:34 raspberrypi python[1240]:     if cb(browser.next()):
May 10 07:53:34 raspberrypi python[1240]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 356, in next
May 10 07:53:34 raspberrypi python[1240]:     raise StopIteration()
May 10 07:53:34 raspberrypi python[1240]: StopIteration
May 10 07:53:34 raspberrypi systemd[1]: screenly-viewer.service: Main process exited, code=exited, status=1/FAILURE
May 10 07:53:34 raspberrypi systemd[1]: screenly-viewer.service: Unit entered failed state.
May 10 07:53:34 raspberrypi systemd[1]: screenly-viewer.service: Failed with result 'exit-code'.

pi@raspberrypi:/var/log $ tail -30 syslog
May 10 07:55:15 raspberrypi python[2113]:     if cb(browser.next()):
May 10 07:55:15 raspberrypi python[2113]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 356, in next
May 10 07:55:15 raspberrypi python[2113]:     raise StopIteration()
May 10 07:55:15 raspberrypi python[2113]: StopIteration
May 10 07:55:15 raspberrypi python[2113]: Traceback (most recent call last):
May 10 07:55:15 raspberrypi python[2113]:   File "/home/pi/screenly/viewer.py", line 458, in <module>
May 10 07:55:15 raspberrypi python[2113]:     main()
May 10 07:55:15 raspberrypi python[2113]:   File "/home/pi/screenly/viewer.py", line 439, in main
May 10 07:55:15 raspberrypi python[2113]:     browser_url(url=url)
May 10 07:55:15 raspberrypi python[2113]:   File "/home/pi/screenly/viewer.py", line 272, in browser_url
May 10 07:55:15 raspberrypi python[2113]:     browser_send('uri ' + escaped_url, cb=cb)
May 10 07:55:15 raspberrypi python[2113]:   File "/home/pi/screenly/viewer.py", line 252, in browser_send
May 10 07:55:15 raspberrypi python[2113]:     load_browser()
May 10 07:55:15 raspberrypi python[2113]:   File "/home/pi/screenly/viewer.py", line 238, in load_browser
May 10 07:55:15 raspberrypi python[2113]:     browser_send(uzbl_rc)
May 10 07:55:15 raspberrypi python[2113]:   File "/home/pi/screenly/viewer.py", line 248, in browser_send
[... 28 lines cut ...]
May 10 07:39:37 raspberrypi NetworkManager[324]: <info>  [1525937977.2564] device (wlan0): supplicant interface state: disabled -> inactive
May 10 07:44:53 raspberrypi NetworkManager[324]: <info>  [1525938293.2339] device (wlan0): set-hw-addr: set MAC address to 6A:40:C6:72:C9:19 (scanning)
May 10 07:44:53 raspberrypi NetworkManager[324]: <info>  [1525938293.2626] device (wlan0): supplicant interface state: inactive -> disabled
May 10 07:44:53 raspberrypi NetworkManager[324]: <info>  [1525938293.2679] device (wlan0): supplicant interface state: disabled -> inactive
May 10 07:50:09 raspberrypi NetworkManager[324]: <info>  [1525938609.1820] device (wlan0): set-hw-addr: set MAC address to 32:02:2B:08:85:6E (scanning)
May 10 07:50:09 raspberrypi NetworkManager[324]: <info>  [1525938609.2037] device (wlan0): supplicant interface state: inactive -> disconnected
May 10 07:50:09 raspberrypi NetworkManager[324]: <info>  [1525938609.2054] device (wlan0): supplicant interface state: disconnected -> inactive
[... cut ...]
May 10 07:56:22 raspberrypi kernel: [1266953.830570] Under-voltage detected! (0x00050005)
May 10 07:57:04 raspberrypi kernel: [1266995.430866] Voltage normalised (0x00000000)
May 10 07:57:06 raspberrypi kernel: [1266997.511082] Under-voltage detected! (0x00050005)
May 10 07:57:37 raspberrypi kernel: [1267028.711146] Voltage normalised (0x00000000)
May 10 07:57:39 raspberrypi kernel: [1267030.791295] Under-voltage detected! (0x00050005)
```

There are undervoltage messages, which is unusual since I have a beefy 3A PSU, but the rest of the messages are interesting...

vpetersson, 2018-05-10:
My guess is that it ran into some kind of condition that spiked CPU/mem usage (probably in uzbl). Are these large images?

JonnyTech, 2018-05-10:
They are tiny png's, here is a listing of the first few dozen:

```
pi@raspberrypi:~/screenly_assets $ ls -l | head -40
total 450908
-rw-r--r-- 1 pi pi      7421 Mar 27 18:00 cfb09cd8ce504a91b83de73f99672a27
-rw-r--r-- 1 pi pi      9882 Mar 27 18:01 bb3b2f55b94a47209695e3dfb3d93859
-rw-r--r-- 1 pi pi     10603 Mar 27 18:00 0cc0e5721e814d4fac82cda24a15c024
-rw-r--r-- 1 pi pi     10612 Mar 27 18:01 3b22bd9b66ef4d179e586535c79db11d
-rw-r--r-- 1 pi pi     11817 Mar 27 18:02 ad9503c464bc41a4b372843517b659c2
-rw-r--r-- 1 pi pi     11885 Mar 27 18:00 686d05f78ac34e7ea600e1dcc4894dec
-rw-r--r-- 1 pi pi     11905 Mar 27 18:02 d9b87143b3064101a666209047d74361
-rw-r--r-- 1 pi pi     13822 Mar 27 18:02 94c7eaeee95144ceb5f61fb42516b46e
-rw-r--r-- 1 pi pi     13927 Mar 27 18:01 e6ccd338fbaa4095861d9d6542db431d
-rw-r--r-- 1 pi pi     16034 Mar 27 18:00 db83a3727f374db9aaf32a69ca383e7d
-rw-r--r-- 1 pi pi     16073 Mar 27 18:00 6aa21a42fdbd411492b92d57de3deadd
-rw-r--r-- 1 pi pi     16215 Mar 27 18:01 f9d529be10a24603894bd9a1ca748eda
-rw-r--r-- 1 pi pi     16245 Mar 27 18:00 5c006d5cf1e6401d936ad104caed0517
-rw-r--r-- 1 pi pi     16671 Mar 27 18:00 fac86f38217f4270a7460fb7491ab4d5
-rw-r--r-- 1 pi pi     17537 Mar 27 18:01 e6e0959592044af78312cdd01d37cea9
-rw-r--r-- 1 pi pi     17981 Mar 27 18:01 9eaf542fb2c141bcb46a934ad59cc556
-rw-r--r-- 1 pi pi     18083 Mar 27 18:00 da34c9897ca84f81b3dec7bb9086e0a5
-rw-r--r-- 1 pi pi     18103 Mar 27 18:01 92d3e65f76cf442e908b708710c4a7c6
-rw-r--r-- 1 pi pi     18153 Mar 27 18:01 323a95a3051c4dc4a5b2f2d165304623
-rw-r--r-- 1 pi pi     18469 Mar 27 18:01 20e3b8b6b9724bc6b671e382ae506623
-rw-r--r-- 1 pi pi     18480 Mar 27 18:00 a0f1823beafd4ef7b412337f889ca4a5
-rw-r--r-- 1 pi pi     18483 Mar 27 18:00 648c4630afec4a5fa55d5f0b130696da
-rw-r--r-- 1 pi pi     18497 Mar 27 18:00 dcf9227e8f434709bc4a1028b8e94797
-rw-r--r-- 1 pi pi     19128 Mar 27 18:00 b88d6262cd164f68824db6c6a68a7893
-rw-r--r-- 1 pi pi     19163 Mar 27 18:02 f731e5f6caf840c79cc8808bfd13af23
-rw-r--r-- 1 pi pi     19252 Mar 27 18:02 faae7199dff2496a94327596a5283850
-rw-r--r-- 1 pi pi     19259 Mar 27 18:01 d2a9da04a8b9477fbfff793565eefd81
-rw-r--r-- 1 pi pi     19470 Mar 27 18:03 1b4c97f47c8b4336b8e1a651019df59a
-rw-r--r-- 1 pi pi     19496 Mar 27 18:00 d2b8f3621b4b46a48c2c20aae3aba647
-rw-r--r-- 1 pi pi     19508 Mar 27 18:02 fe49f81ba3f6412dbe5a24670257c934
-rw-r--r-- 1 pi pi     19543 Mar 27 18:00 5f1108805fef4b30a69ce1701add4837
-rw-r--r-- 1 pi pi     19641 Mar 27 18:01 d429d9330bf346c399779832a3be18fc
-rw-r--r-- 1 pi pi     19976 Mar 27 18:00 f6b52adf6455484698494a7803f0a846
-rw-r--r-- 1 pi pi     19986 Mar 27 18:02 19c94dfaa8654e909ccbf545a0c378a8
-rw-r--r-- 1 pi pi     20248 Mar 27 18:00 c1f1eda5d6ad41558889057de094e73e
-rw-r--r-- 1 pi pi     20301 Mar 27 18:01 dd3e91d7b7b848028fdfec55bd41aba2
-rw-r--r-- 1 pi pi     20304 Mar 27 18:01 79cd2d3d30d843b38e59a1db7cd181b0
-rw-r--r-- 1 pi pi     20312 Mar 27 18:01 9554c3996cce4f0d8cbb0e0632edba3f
-rw-r--r-- 1 pi pi     20640 Mar 27 18:00 d74c6cb6d04b4ae09bb08bd068f5a9db
```

vpetersson, 2018-05-10:
Just as an experiment - could you try doubling the duration. Also, I wouldn't rule out the power issue. Even if your PSU is marked 3A, lots of them actually fail to deliver that consistently. This is fine for mot user cases (such as phones), but can cause all sorts of random issues on something like the Raspberry Pi.

JonnyTech, 2018-05-10:
OK, I shall try that shortly - unless there is anything else that you would like me to check before rebooting it.

The [official recommendation](https://www.raspberrypi.org/help/faqs/#powerReqs) for the Raspberry Pi 3 Model B is 2.5A so to avoid any problems I got higher specced dedicated units from a [reputable supplier](https://www.robotshop.com/uk/raspberry-pi-power-supply-5v-3a-micro-usb.html).

I have also tried with [other decent](https://www.anker.com/store/PowerPort-6-Ports/A2123123) chargers but still get occasional lock-ups.

JonnyTech, 2018-05-10:
Eek - is there an easy way to change the timings of multiple assets? Manually changing 100 files is tedious!

antonmolodykh, 2018-05-10:
you can do it manually or using API. 
By the way: are you sure that you are using Pi 3 Model B? I started receiving notifications like `Under-voltage detected! (0x00050005)` only when I switched to Pi 3 Model B+ (it requires 3A)

vpetersson, 2018-05-10:
> OK, I shall try that shortly - unless there is anything else that you would like me to check before rebooting it.

Yes, please check the memory (`free -m`) and cpu usage (`uptime`).

> The official recommendation for the Raspberry Pi 3 Model B is 2.5A so to avoid any problems I got higher specced dedicated units from a reputable supplier.

I'm not familiar with this particular one, but we always recommend the official PSU as it is made to the specs.

JonnyTech, 2018-05-10:
@antonmolodykh my bad, you're right, it's a 3B+. Where does it document a 3A requirement? The [official blog post](https://www.raspberrypi.org/blog/pi-power-supply-chip/) recommends their [official PSU](https://www.raspberrypi.org/products/raspberry-pi-universal-power-supply/) which is rated at 2.5A. I ordered them but when two died within a week, I replaced them and had no more failures (or yellow lightning bolt symbols). 

> you can do it manually or using API

Do you have an absolute beginners guide to using the API?

@vpetersson sorry, too late, I had to get it running again as it was required.

antonmolodykh, 2018-05-10:
@JonnyTech oh, now it's my bad, you're right, really 2.5A for 3b+.
We don't have a guide, but a full description of the api is available at `/api/docs` address (`http://your_raspberry_ip/api/docs`).

JonnyTech, 2018-05-10:
@antonmolodykh `/api/docs` = wow, I did not know that existed! I may start a new thread with questions....

closed 2018-10-19 by stale[bot]