Screenly/Anthias #787: Docker-Image not work

xstable, 2018-04-06:
If I try to start the Docker-Image with this command: 
`docker run --rm -ti --name=screenly-dev -e 'LISTEN=0.0.0.0' -p 8080:8080 -v $(pwd):/home/pi/screenly screenly/ose-dev-server`
I got this response: 
`docker: Error response from daemon: failed to create endpoint screenly-dev on network bridge: failed to add the host`

How can I fix this?

xstable, 2018-04-06:
Great, I got it. Thought the screenly-ose docker-image was a full packed container with sources inside. 
Now I've download the sources and run the container... works like a charm.
Thank you.

closed 2018-04-06 by xstable

vpetersson, 2018-04-06:
> Thought the screenly-ose docker-image was a full packed container with sources inside.

Actually, it is, but it wouldn't make it a very good development environment if you need to edit files inside the docker container.