Screenly/Anthias #788: Where to see output

xstable, 2018-04-06:
I run Screenly in docker-container and can add assets to it.
All seems to work like a charm, but I wonder where to see the output of those assets.

vpetersson, 2018-04-06:
You mean as a simulated screen? If so, you can't. The docker environment only covers the servers side, not the viewer.

xstable, 2018-04-06:
Ok, what is the way the viewer works by default? This only work on Raspery-Pi, or am I able to get some output at my dev-pc?

My enviroment
==
- php-script lay on the server
- screenly-ose run on my local machine (docker container)

What I try to do: 
==
- Customer has an Raspery-PI with Screenly (I don't have access to it).
- I should create an external php-file which fetch some data and do some animation
- because animation has different (unknown) durations, the php-script should trigger the api of screenly if animation is finisehd - to swtich to next asset.

How can I figure out if my php-script is displayed well in screenly?

vpetersson, 2018-04-06:
> This only work on Raspery-Pi, or am I able to get some output at my dev-pc?

It should be possible to run the viewer in development environment on a Linux desktop (I know @antonmolodykh does).

> I should create an external php-file which fetch some data and do some animation

Let me stop you there. This is unlikely to work well. The Raspberry Pi is not powerful enough to handle HTML animations (at least with the current browser). Your only hope is #527 which incidentally @antonmolodykh is making good progress on as we speak.

If you want animations to work well, your best option is either to encode them into a video, or switch to Screenly Pro.

xstable, 2018-04-06:
Thank you for your Feedback. 
I use Linux, so glad to hear, there'll maybe a way to run the viewer. Did you have a link, how to do this?

The other Point: Animation was the wrong description. To be more precise, I fetch single Youtube-Videos out of a given Playlist. 
So the user only have to edit and sort a given playlist to change the Videos and their sort order inside his Screenly-Animation.
**Reason**: The User don't wan't to edit the Video-URL on each asset.

xstable, 2018-04-06:
So what if you go inside the docker-container with _-ti .... /bin/bash_ and there run server.py and (maybe with the CLI-Programm "screen" the "viewer.py"? [BTW: what is the root-password inside docker? I need it to install screen, to give it a try]

_What are the default Steps on Raspery-Pi?_ 

1. Install screenly on raspery-pi 
2. run the server.py (?)
3. run the viewer.py (?)

Would be fine to have a little HowTo at the page which makes developers able to setup such a small dev-enviroment to get faster results with screenly.

vpetersson, 2018-04-06:
No, you won't be able to run the viewer inside of the Docker container (since it need to talk to X11). 

> what is the root-password inside docker? 

It's an upstream image. Normally you don't run screen etc inside a docker container. They're designed to run only a single process.

You're probably better off trying to get it all running stand-alone.

vpetersson, 2018-04-10:
Nice find @xstable. Would be great if you can produce a `Vagrantfile` that does all of that.

xstable, 2018-04-10:
Sorry, not yet familiar with vagrant. Have created it in Oracle Vbox.