Screenly/Anthias #789: Error loading google presentation with youtube videos

nrdroque, 2018-04-06:
hi i'm testing your version OSE to run google presentations, and some of them need to have youtube videos inserted in some context.
but when loading the youtube it gives me error 5.
i've tested the google presentation over several browsers and computer settings and it work well

vpetersson, 2018-04-06:
Yes, that will not work. In short, the Raspberry Pi is not powerful enough to do this (at least not using the current browser). Keep an eye on #527 as that could potentially change that.

closed 2018-04-06 by vpetersson

antonmolodykh, 2018-06-20:
@nrdroque we published the experimental screenly ose version. one of the major changes is the new browser. To try it follow the steps under 'Manual upgrade' [here](https://www.screenly.io/ose/) and answer yes to the question about experimental version.
We are looking forward to feedback, please let us know about any problems.