Screenly/Anthias #790: Assets not refreshing - Google Slides

Freyli, 2018-04-06:
Changes are not being updated when made in Google Sildes. Clicking the asset link opens up to the latest page. Web page asset not reloading. I know this issue had been brought up in 2016 but it doesn't appear to be working at this time.

vpetersson, 2018-04-06:
What duration did you set? It will not reload before the duration expires.

JaapJonker, 2018-05-28:
Hi there,

I've the same issue. I have only one asset and the time is set on 10 seconds. 

Thanks for helping out!

Jaap

JaapJonker, 2018-05-28:
@Freyli For the moment I have a work around:

1. Make two assets with the links of your Google Slide, each with a slightly different time delay (I have a 10000ms and a 10001 ms link.
2. Set the duration of the assets equally as your number of slides * duration in seconds in Google Slides.

This will reload the Google Slides Presentation every time the presentation starts over.

vpetersson, 2018-08-13:
@Freyli Could you try the experimental branch?

closed 2018-10-19 by stale[bot]