Screenly/Anthias #792: Auto login and element hiding

mattjford, 2018-04-08:
For Advanced Users:

Could there be a auto login (support for Google ideally) as well as element hiding.  

Today I use selenium to handle most of my signage needs and I am able to manipulate the DOM fairly easily.  Would that be a possibility?

Due to iframe cross origin issues it's near impossible for me to simply add a JS layer on top of an iframe to perform the same.  It would almost need to be a module extension for the same, and that alters your core viewer.

vpetersson, 2018-08-13:
(Sorry for the late response. Missed this one.)

@mattjford That's a very cool feature. However, to my knowledge, we can't really run Selenium inside either the old browser (uzbl) or new browser (qt-webkit) since it isn't supported there.

closed 2018-10-19 by stale[bot]

mattjford, 2018-10-19:
Not sure if you'll see this, but I wonder if you could add a js wrapper
that could do most of it instead of relying on selenium

mattjford, 2018-10-19:
XSS would still be an issue though, nevermind