Screenly/Anthias #794: API error file_upload

peter-4A5A, 2018-04-11:
Hi everyone,

When I am trying to add a file asset trough the screenly api, I always get a error 500 internal server error.

My post request that I send i as following:
`array(1) { ["file_upload"]=> string(17) "_MON2760kopie.jpg" }`

The error log that i get in my console is:
`Apr 11 10:56:23 raspberrypi python[530]:   File "/home/pi/screenly/server.py", line 238, in api_view
Apr 11 10:56:23 raspberrypi python[530]:     try:
Apr 11 10:56:23 raspberrypi python[530]:   File "/usr/local/lib/python2.7/dist-packages/flask_restful_swagger_2/swagger.py", line 219, in inner
Apr 11 10:56:23 raspberrypi python[530]:     return f(self, *args, **kwargs)
Apr 11 10:56:23 raspberrypi python[530]:   File "/home/pi/screenly/server.py", line 550, in post
Apr 11 10:56:23 raspberrypi python[530]:     logging.info('import', req)
Apr 11 10:56:23 raspberrypi python[530]: AttributeError: 'NoneType' object has no attribute 'filename'`

The asset that i want to add is uploaded on the following location: /home/pi/screenly_assets

Under the name: _MON2760kopie.jpg.tmp

Could someone help me and explain what I am doing wrong?

Thank you,

antonmolodykh, 2018-04-11:
I was able to get this error only if I tried to load a file as binary data. A file should be send as form-data, the field should be named "file_upload".
However, you need to remember that this endpoint serves to send a file, and not to add an asset. That is, if you want to add a file as an asset, then you need 
* upload your file to the Pi `POST /api/v1/file_asset` (the response contains the current file **uri**) 
* add an asset `POST /api/v1.1/assets` using the file uri

peter-4A5A, 2018-04-13:
Thank you for your feedback.

I am trying to upload the file to my screenly using the following PHP code:
`      $request = curl_init('http://x.x.x.x/api/v1/file_asset');
      // send a file
      curl_setopt($request, CURLOPT_POST, true);
      curl_setopt(
          $request,
          CURLOPT_POSTFIELDS,
          array(
            'file_upload' =>
                '@'            . $_FILES['file_upload']['tmp_name']
                . ';filename=' . $_FILES['file_upload']['name']
                . ';type='     . $_FILES['file_upload']['type']
          ));

      // output the response
      curl_setopt($request, CURLOPT_RETURNTRANSFER, true);
      $uploadResult = curl_exec($request);
      var_dump($uploadResult);
      // close the session
      curl_close($request);`

You can see a example of how this work here: http://code.iamkate.com/php/sending-files-using-curl/

And this should replicate a form-data post request

antonmolodykh, 2018-04-16:
@peter-4A5A unfortunately I've never worked with PHP and I cannot tell you what the problem is, or how it's done correctly in this language. However you can be sure that these endpoints work as I described if you read our js code or try to use Postman.

peter-4A5A, 2018-04-16:
@antonmolodykh thank you for your response.

I will search for a solution and after i found one, I will let you know for anyone in the further who might has this problem.

closed 2018-04-16 by peter-4A5A

peter-4A5A, 2018-06-11:
I solved the issue using a curl library: https://github.com/php-curl-class/php-curl-class