Screenly/Anthias #795: Videos smooth but webpage animations very choppy/laggy

jeffdesc, 2018-04-12:
Installed Screenly OSE on a Raspbian Stretch Lite as the last image does not work. However the videos are very smooth and have no issue with at all. But webpages with some animations in it are very choppy or laggy, it is not due to the network connection (gigabit Ethernet).

For instance we tried to show some [attack maps](http://map.norsecorp.com/), but it is very laggy. Instead of fluent motions it's a 3 fps slideshow.

Can this be improved due to some Raspbian tweaks/OSE tweaks. Are there some roadmaps to use another viewer as UZBL to improve this kind of pages?

vpetersson, 2018-04-12:
Yep, see #527. We're making good progress on this and will need testers for it in the near future.

closed 2018-04-12 by vpetersson

jeffdesc, 2018-04-12:
@vpetersson Thanks for your fast confirmation. Will this be tested in a closed environment or can we test it as well? Would be happy to take part in it.

vpetersson, 2018-04-12:
At the moment, we don't have anything to test. @antonmolodykh is working on trying to wrap up the last parts. Once done, we'll most likely create an experimental branch that will use the browser, which will allow users to manually opt in and test.

jeffdesc, 2018-04-12:
Thanks @vpetersson!