Screenly/Anthias #796: Running without network on 2018-01-19-Screenly_OSE

BananaAcid, 2018-04-12:
After boot, when the first of the images should start without network cable on RPI1B,
it just shows a gray background and the console login.

Plugging in the cable and rebooting works fine.

Using 2018-01-19-Screenly_OSE_4GB.img

vpetersson, 2018-04-12:
That's strange. We've made some changes to this. Could you try upgrading to the latest development version to see if that mitigates the problem?

BananaAcid, 2018-04-12:
How do I do this?

vpetersson, 2018-04-12:
Please follow the steps under 'Manual upgrade' [here](https://www.screenly.io/ose/) and answer yes to the question about the development version.

BananaAcid, 2018-04-12:
reboot: console is back..

antonmolodykh, 2018-04-18:
I think this related to pi1. I'll check and let you know

vpetersson, 2018-06-22:
@BananaAcid I *think* this might be related to #836. We will be adding more diverse hardware to the testing platform in the near future. Closing out this one for now however.

closed 2018-06-22 by vpetersson