Screenly/Anthias #797: Assets disabled after update

morituro, 2018-04-12:
hi folks,
thx for your fantastic programm. we're using it at our school and it works fine. but after updating screenly with a bash command all assets were disabled and i had to upload them once again. i had the same issue when we gave the pi a new static ip...
thx for your help!

antonmolodykh, 2018-04-18:
Hi @morituro 
what do you mean "all assets were disabled"? assets completely disappeared from the web interface?

morituro, 2018-04-18:
thx for your response! i mean they are deactivated - button off.

antonmolodykh, 2018-04-19:
hmm. if the device has no problems with time, an asset should be active if:
* user did not click "off" in the right column
* asset has not expired (see "start" and "end" columns).
are both conditions met?

closed 2018-05-04 by vpetersson