Screenly/Anthias #799: screenly-ose works on  Pi 2 Model B v1.1 but not on PI 3 Model B

dudul08, 2018-04-17:


vpetersson, 2018-04-17:
@dudul08 Can you be more specific? We have lots of users on Pi 3 Model B so I know that it works.

dudul08, 2018-04-17:
On boot, the screen is fixed on a "rainbow"...

vpetersson, 2018-04-17:
Ah! You're talking about the *Raspberry Pi 3 Model B+*. That's a different board. Yes, that is a known issue at the moment. You will have to do the manual installation for now.

dudul08, 2018-04-17:
Ok !
Thank you for help !

closed 2018-04-17 by dudul08

victorehi, 2018-09-18:
Hey @vpetersson, 
When you say Manual Installation for the Raspberry Pi 3 Model B+ and Screenly Pro 2, what do you mean?

vpetersson, 2018-09-18:
@victorehi I'm not sure this is relevant to this current issue, but take a look at [Screenly Pro vs OSE](https://www.screenly.io/screenly-ose-vs-pro/) for the difference between OSE and Pro as well as the Option 3 on the [OSE page](https://www.screenly.io/ose/).