Screenly/Anthias #801: White flickering screen

Zer0x17, 2018-04-18:
Hi,
After I completed my project with the latest sprint of screenly ose and the system started up, only the screen was flickering white. No content was displayed. Assets are included.

Following the log files.
`Started User Manager for UID 1000.
Apr 18 19:00:54 raspberrypi python[769]: browser found dead, restarting
Apr 18 19:00:54 raspberrypi python[769]: Loading browser...
Apr 18 19:00:54 raspberrypi python[769]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 798.
Apr 18 19:00:54 raspberrypi python[769]: current url is file:///tmp/screenly_html/black_page.html
Apr 18 19:00:54 raspberrypi python[769]: Scheduler init
Apr 18 19:00:54 raspberrypi python[769]: update_playlist
Apr 18 19:00:54 raspberrypi python[769]: Generating asset-list...
Apr 18 19:00:54 raspberrypi python[769]: generate_asset_list deadline: 2018-03-14 17:47:00
Apr 18 19:00:54 raspberrypi python[769]: update_playlist done, count 4, counter 0, index 0, deadline 2018-03-14 17:47:00
Apr 18 19:00:54 raspberrypi python[769]: Entering infinite loop.
Apr 18 19:00:54 raspberrypi python[769]: Last update: 2018-04-16 10:38:56.989833
Apr 18 19:00:54 raspberrypi python[769]: Viewer crashed.
Apr 18 19:00:54 raspberrypi python[769]: Traceback (most recent call last):
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
Apr 18 19:00:54 raspberrypi python[769]:     main()
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 452, in main
Apr 18 19:00:54 raspberrypi python[769]:     asset_loop(scheduler)
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 373, in asset_loop
Apr 18 19:00:54 raspberrypi python[769]:     check_update()
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 343, in check_update
Apr 18 19:00:54 raspberrypi python[769]:     'Hash': str(git_hash),
Apr 18 19:00:54 raspberrypi python[769]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 95, in track
Apr 18 19:00:54 raspberrypi python[769]:     self._consumer.send('events', json_dumps(event, cls=self._serializer))
Apr 18 19:00:54 raspberrypi python[769]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 353, in send
Apr 18 19:00:54 raspberrypi python[769]:     self._write_request(self._endpoints[endpoint], json_message, api_key)
Apr 18 19:00:54 raspberrypi python[769]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 376, in _write_request
Apr 18 19:00:54 raspberrypi python[769]:     six.raise_from(MixpanelException(e), e)
Apr 18 19:00:54 raspberrypi python[769]: AttributeError: 'module' object has no attribute 'raise_from'
Apr 18 19:00:54 raspberrypi python[769]: Traceback (most recent call last):
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
Apr 18 19:00:54 raspberrypi python[769]:     main()
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 452, in main
Apr 18 19:00:54 raspberrypi python[769]:     asset_loop(scheduler)
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 373, in asset_loop
Apr 18 19:00:54 raspberrypi python[769]:     check_update()
Apr 18 19:00:54 raspberrypi python[769]:   File "/home/pi/screenly/viewer.py", line 343, in check_update
Apr 18 19:00:54 raspberrypi python[769]:     'Hash': str(git_hash),
Apr 18 19:00:54 raspberrypi python[769]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 95, in track
Apr 18 19:00:54 raspberrypi python[769]:     self._consumer.send('events', json_dumps(event, cls=self._serializer))
Apr 18 19:00:54 raspberrypi python[769]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 353, in send
Apr 18 19:00:54 raspberrypi python[769]:     self._write_request(self._endpoints[endpoint], json_message, api_key)
Apr 18 19:00:54 raspberrypi python[769]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 376, in _write_request
Apr 18 19:00:54 raspberrypi python[769]:     six.raise_from(MixpanelException(e), e)
Apr 18 19:00:54 raspberrypi python[769]: AttributeError: 'module' object has no attribute 'raise_from'
Apr 18 19:00:55 raspberrypi systemd[1]: screenly-viewer.service: Main process exited, code=exited, status=1/FAILURE
Apr 18 19:00:55 raspberrypi systemd[1]: screenly-viewer.service: Unit entered failed state.
Apr 18 19:00:55 raspberrypi systemd[1]: screenly-viewer.service: Failed with result 'exit-code'.
Apr 18 19:00:55 raspberrypi systemd[1]: screenly-viewer.service: Service hold-off time over, scheduling restart.
Apr 18 19:00:55 raspberrypi systemd[1]: Stopped Screenly Viewer.
Apr 18 19:00:55 raspberrypi systemd[1]: Starting Screenly Viewer...
Apr 18 19:00:55 raspberrypi systemd[1]: Started Screenly Viewer.
Apr 18 19:00:56 raspberrypi python[831]: browser found dead, restarting
Apr 18 19:00:56 raspberrypi python[831]: Loading browser...
Apr 18 19:00:56 raspberrypi python[831]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 843.
Apr 18 19:00:57 raspberrypi python[831]: current url is file:///tmp/screenly_html/black_page.html
Apr 18 19:00:57 raspberrypi python[831]: Scheduler init
Apr 18 19:00:57 raspberrypi python[831]: update_playlist
Apr 18 19:00:57 raspberrypi python[831]: Generating asset-list...
Apr 18 19:00:57 raspberrypi python[831]: generate_asset_list deadline: 2018-03-14 17:47:00
Apr 18 19:00:57 raspberrypi python[831]: update_playlist done, count 4, counter 0, index 0, deadline 2018-03-14 17:47:00
Apr 18 19:00:57 raspberrypi python[831]: Entering infinite loop.
Apr 18 19:00:57 raspberrypi python[831]: Last update: 2018-04-16 10:38:56.989833
Apr 18 19:00:57 raspberrypi python[831]: Viewer crashed.
Apr 18 19:00:57 raspberrypi python[831]: Traceback (most recent call last):
Apr 18 19:00:57 raspberrypi python[831]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
[... 13 lines cut ...]
Apr 18 19:00:57 raspberrypi python[831]: AttributeError: 'module' object has no attribute 'raise_from'
Apr 18 19:00:57 raspberrypi python[831]: Traceback (most recent call last):
Apr 18 19:00:57 raspberrypi python[831]:   File "/home/pi/screenly/viewer.py", line 457, in <module>
Apr 18 19:00:57 raspberrypi python[831]:     main()
Apr 18 19:00:57 raspberrypi python[831]:   File "/home/pi/screenly/viewer.py", line 452, in main
Apr 18 19:00:57 raspberrypi python[831]:     asset_loop(scheduler)
Apr 18 19:00:57 raspberrypi python[831]:   File "/home/pi/screenly/viewer.py", line 373, in asset_loop
Apr 18 19:00:57 raspberrypi python[831]:     check_update()
Apr 18 19:00:57 raspberrypi python[831]:   File "/home/pi/screenly/viewer.py", line 343, in check_update
Apr 18 19:00:57 raspberrypi python[831]:     'Hash': str(git_hash),
Apr 18 19:00:57 raspberrypi python[831]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 95, in track
Apr 18 19:00:57 raspberrypi python[831]:     self._consumer.send('events', json_dumps(event, cls=self._serializer))
Apr 18 19:00:57 raspberrypi python[831]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 353, in send
Apr 18 19:00:57 raspberrypi python[831]:     self._write_request(self._endpoints[endpoint], json_message, api_key)
Apr 18 19:00:57 raspberrypi python[831]:   File "/usr/local/lib/python2.7/dist-packages/mixpanel/__init__.py", line 376, in _write_request
Apr 18 19:00:57 raspberrypi python[831]:     six.raise_from(MixpanelException(e), e)
Apr 18 19:00:57 raspberrypi python[831]: AttributeError: 'module' object has no attribute 'raise_from'
Apr 18 19:00:57 raspberrypi systemd[1]: screenly-viewer.service: Main process exited, code=exited, status=1/FAILURE
Apr 18 19:00:57 raspberrypi systemd[1]: screenly-viewer.service: Unit entered failed state.
Apr 18 19:00:57 raspberrypi systemd[1]: screenly-viewer.service: Failed with result 'exit-code'.
Apr 18 19:00:57 raspberrypi systemd[1]: screenly-viewer.service: Service hold-off time over, scheduling restart.
Apr 18 19:00:57 raspberrypi systemd[1]: Stopped Screenly Viewer.
Apr 18 19:00:57 raspberrypi systemd[1]: Starting Screenly Viewer...
Apr 18 19:00:57 raspberrypi systemd[1]: Started Screenly Viewer.
`

antonmolodykh, 2018-04-19:
Judging by the logs, there is no connection on your device and therefore a known problem arose. We have already solved it in the development branch. To get this, please run ` bash <(curl -sL https://www.screenly.io/install-ose.sh)` and answer yes to the question about the development version.

Zer0x17, 2018-04-20:
Thanks this helped! Very good work by the way! I ♥ it!

closed 2018-04-20 by Zer0x17