Screenly/Anthias #804: How to wait for PUT till assert is inserted

xstable, 2018-04-20:
I have an php-Script which do some PUT-Request.
Inserting some Assets  works well, but I have trouble to order them.

Currently I request all screenly-asserts and then iterate with a modulus operator to insert the video at each second assert.
I've choosed even numbers, so the first video should start with 2, followed by 4,6,8 and so on.

The issue is, that all videos will paste at Position 2... means 2,3,4.5 and then the other pre-existing assets appear.

So here some JSON-Data which I've send and receive: 
```JSON
{
                          "is_enabled": 1,
                          "mimetype": "webpage",
                          "end_date": "2119-04-19T14:54:23.623Z",
                          "is_active": true,
                          "duration": 303,
                          "is_processing": 0,
                          "asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai4yODlGNEE0NkRGMEEzMEQy",
                          "name": "youtube-How to Make a Macro Lens",
                          "nocache": 0,
                          "uri": "https://youtu.be/NNp-tne9KaE",
                          "play_order": "1",
                          "start_date": "2018-04-19T14:54:23.623Z"
                          }
                          
{
                          "is_enabled": 1,
                          "mimetype": "webpage",
                          "end_date": "2119-04-19T14:54:23.623Z",
                          "is_active": true,
                          "duration": 536,
                          "is_processing": 0,
                          "asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai4wOTA3OTZBNzVEMTUzOTMy",
                          "name": "youtube-Holz mit Kaffee, Essigessenz und Stahlwolle verwittern/altern lassen/Treibholz-Effekt | Tutorial/DIY",
                          "nocache": 0,
                          "uri": "https://youtu.be/IHIKlmG-UeM",
                          "play_order": "3",
                          "start_date": "2018-04-19T14:54:23.623Z"
                          }
                          
{
                          "is_enabled": 1,
                          "mimetype": "webpage",
                          "end_date": "2119-04-19T14:54:23.623Z",
                          "is_active": true,
                          "duration": 343,
                          "is_processing": 0,
                          "asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai41MjE1MkI0OTQ2QzJGNzNG",
                          "name": "youtube-40 km/h DIY Electric Longboard",
                          "nocache": 0,
                          "uri": "https://youtu.be/SL1hw6wYEOo",
                          "play_order": "5",
                          "start_date": "2018-04-19T14:54:23.623Z"
                          }
                          
{
                          "is_enabled": 1,
                          "mimetype": "webpage",
                          "end_date": "2119-04-19T14:54:23.623Z",
                          "is_active": true,
                          "duration": 1128,
                          "is_processing": 0,
                          "asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai4xMkVGQjNCMUM1N0RFNEUx",
                          "name": "youtube-Transform a laptop into a stunning desktop media PC (for CHEAP)",
                          "nocache": 0,
                          "uri": "https://youtu.be/e3fnsGHe8eE",
                          "play_order": "7",
                          "start_date": "2018-04-19T14:54:23.623Z"
                          }
```

The Result for the POST-Curl which adds an asset (`play_item` is missing... why???)
```JSON
{
"mimetype": "webpage", 
"asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai4yODlGNEE0NkRGMEEzMEQy", 
"end_date": "2119-04-19T14:54:23.623000+00:00", 
"name": "youtube-How to Make a Macro Lens", 
"nocache": 0,
"is_enabled": 1,
"is_active": true, 
"uri": "https://youtu.be/NNp-tne9KaE", 
"duration": 303, 
"start_date": "2018-04-19T14:54:23.623000+00:00", 
"is_processing": 0}
{
"mimetype": "webpage",
 "asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai4wOTA3OTZBNzVEMTUzOTMy",
 "end_date": "2119-04-19T14:54:23.623000+00:00",
 "name": "youtube-Holz mit Kaffee,
 Essigessenz und Stahlwolle verwittern/altern lassen/Treibholz-Effekt | Tutorial/DIY",
 "nocache": 0,
 "is_enabled": 1,
 "is_active": true,
 "uri": "https://youtu.be/IHIKlmG-UeM",
 "duration": 536,
 "start_date": "2018-04-19T14:54:23.623000+00:00",
 "is_processing": 0
}
{
"mimetype": "webpage",
 "asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai41MjE1MkI0OTQ2QzJGNzNG",
 "end_date": "2119-04-19T14:54:23.623000+00:00",
 "name": "youtube-40 km/h DIY Electric Longboard",
 "nocache": 0,
 "is_enabled": 1,
 "is_active": true,
 "uri": "https://youtu.be/SL1hw6wYEOo",
 "duration": 343,
 "start_date": "2018-04-19T14:54:23.623000+00:00",
 "is_processing": 0
}
{
"mimetype": "webpage",
 "asset_id": "UExUNFNHYmxjRGUtcnVBZHJMNnJiM3VpWkRUS0o4M1E0ai4xMkVGQjNCMUM1N0RFNEUx",
 "end_date": "2119-04-19T14:54:23.623000+00:00",
 "name": "youtube-Transform a laptop into a stunning desktop media PC (for CHEAP)",
 "nocache": 0,
 "is_enabled": 1,
 "is_active": true,
 "uri": "https://youtu.be/e3fnsGHe8eE",
 "duration": 1128,
 "start_date": "2018-04-19T14:54:23.623000+00:00",
 "is_processing": 0
}
```

And here the result... the red-borderd assets are the Videos insert by POST: 

[image]

antonmolodykh, 2018-04-21:
I can not test this right now, however I suppose that problems can arise because of the wrong type (play_order must be int)

xstable, 2018-04-23:
And here is a Video to reproduce the BUG: http://somup.com/cFf02KV5CI

antonmolodykh, 2018-04-23:
Thank you @xstable  
I'll check it tomorrow and let you know

xstable, 2018-04-27:
@antonmolodykh Did you figured out if this is a bug?

antonmolodykh, 2018-04-27:
@xstable sorry for the wait. 
I refreshed this in my memory: this is the expected behavior. To change the order of assets there is a `/api/v1/assets/order` endpoint. 
A detailed description of the endpoints is available at the `api/docs` address

xstable, 2018-04-27:
Ok, but why does the model has an _play_order_-Field in the model, if it is not used. 
Anyway, it would make things much, much easier if you could set this directly in your POST-Request.
It's absolutely logic and straight forward to say which Position the new Asset should have been, without doing an additional API-request.

Currently I add a bunch of videos (from Youtube-Playlist) in a batch-job where each 2nd video should be added via API-POST. Screenly take this POST-REQUEST and proceed with the Video-Download.

So first big Problem here is, that you get a response if the youtube-process is started, but not if it is finished ( means Video downloaded and activated). So while you got the return-value that the POST succeed, screenly go on to proceed with:

- Download the Video
- Rename the Title to the Name fetch from Youtube-Video
- set the is_processing-Flag to '0'

Because of this: 
- Your new added Video is in the asset_list and looks like each other!
- You don't be able to create a chain with the response of the POST-Request. While screenly proceed in background to doing the above described Steps, while PHP-Script go on to POST the Next-Video from the Batch-Process. 

Even downsides which make sorting Assets hard:  
- You don't can use a Placeholder in the Name to mark those new Videos (e.g. Youtube_), because Asset-Name is changed after Video is fully downladed
- The asset-type is set to Video, which don't differ this Video from pre-existing Videos in the Asset-List

The only way to solve this would be to:

1. Fetch all Assets and save the result on runtime (e.g. PHP-Object)
2. Fetch your Videolist and save the result on runtime (e.g. PHP-Object)
3. POST your Videos (which place Videos up from pos2)
4. Fetch all Assets again (only to resort them all ... which means):
   Parse Asset-Object, compare to Videolist-Object, find each added Video and rearrange it to the order you need (huge logic, if you don't has a static video-list)

You see, this is a very _huge downside_. It could be so simple and logic, if you only make _play_order_ work for a POST-API-Call. If would prevent to have such a huge effort like described above!

 
Please, please, please...  make _play_order_ works for API-POST-Request (adding new Assets). This would make it much easier to put new Assets at the right place.

xstable, 2018-04-27:
**BTW:** What will it cost, If you add this adjustment (maybe a patched file) specially for my needs?
This way I could only patch my screenly-systems to make POST-Request using _play_order_.