Screenly/Anthias #806: Grafana graphs not loading on Screenly OSE

jeffdesc, 2018-04-23:
We are using Grafana to display some data. As we're using Screenly and surprised by it, we thought to put Grafana on Screenly as well.

However when loading the Grafana snapshot we're only seeing the Grafana layout and the rows. The graphs themselfs are not being rendered at all.

We're using the lastest Screenly version and Grafana v5.0.4.

Is there any workaround for this?

antonmolodykh, 2018-04-24:
I think this is related to uzbl, which we use to display web pages. Soon we will release an update with a new webview.

jeffdesc, 2018-04-30:
Thanks @antonmolodykh. Is there any ETA available yet for the new webview (or atleast in testing/developer stage)? Are we speaking about days, weeks or months?

antonmolodykh, 2018-04-30:
Webview at the development stage, but right now we have several more priority tasks. I think that we are speaking about weeks.

vpetersson, 2018-08-10:
@jeffdesc Forgot to update you on this, but could you try the latest experimental branch? It has the new browser.

jeffdesc, 2018-08-16:
@vpetersson Thanks for the update. Can I just update my existing installation or do you want me rather to use a fresh install?

vpetersson, 2018-08-16:
You could upgrade the existing one to the experimental branch

Minimerl, 2018-08-17:
Had a similar problem earlier today where Screenly displayed the legend but not the graph
Setting a custom width on the legend in Grafana did the trick

vpetersson, 2018-08-20:
Interesting workaround @Minimerl. 

@jeffdesc The root cause is likely that the version is of webkit in the production branch of OSE is rather dated by now unfortunately. We are working on this in the experimental version.

jeffdesc, 2018-09-10:
@vpetersson sorry for my late answer. I am busy at the moment on my job. Is the last issue fixed already and I can test this or do you want me to wait a little longer before testing?

vpetersson, 2018-09-10:
@jeffdesc Feel free to test whenever you got time.

RaiderXXL, 2018-11-07:
Have an other problem that grafana site is just black... nothing is displayed... 
Access to grafana is without athentication. Can connect with all other devices/browsers. 
Did i missed somthing? 

Have updated Screenly today to Development version.

vpetersson, 2018-11-12:
@RaiderXXL Interesting. Can you verify that caching is enabled in the experimental browser?

RaiderXXL, 2018-11-12:
Hello, How can i do this? Is there a Wiki or help file?

vpetersson, 2018-11-12:
@RaiderXXL Sorry, that was a comment for @rusko124

RaiderXXL, 2018-11-22:
Still there .. waiting... Blackscreen still persist.
Any news for me?

vpetersson, 2018-11-22:
@rusko124 - i guess we need to update the experimental branch for the new release. Can you look at this today?

rusko124, 2018-11-23:
@RaiderXXL Can you update your experimental branch and tell me if it helped you, please?

RaiderXXL, 2018-11-23:
Update done reboot done black screen still there...

Update via "bash <(curl -sL https://www.screenly.io/install-ose.sh)" 
choose experimental

[new 7.txt](https://github.com/Screenly/screenly-ose/files/2609949/new.7.txt)

RaiderXXL, 2018-12-10:
nothing happening here?

vpetersson, 2018-12-10:
@RaiderXXL it's in the backlog for the next sprint, but we welcome code contributions to accelerate it.

DuschdrBabbe, 2019-02-13:
Dont close this Tonic, please

ealmonte32, 2019-03-13:
@DuschdrBabbe @jeffdesc 

Running production version of screenly-ose and this is what I get when I visit the Grafana demo page:
[image]

Screenly Pi running:
```
pi@rp300we:~ $ uname -a
Linux rp300we 4.19.25-v7+ #1205 SMP Mon Feb 25 18:19:20 GMT 2019 armv7l GNU/Linux
pi@rp300we:~ $ dpkg -l | grep web
ii  libwebkit2gtk-4.0-37:armhf           2.18.6-1~deb9u1+rpi1              armhf        Web content engine library for GTK+
ii  libwebkitgtk-1.0-0:armhf             2.4.11-3+rpi1                     armhf        Web content engine library for GTK+
```

ealmonte32, 2019-03-13:
Pi was imaged with Raspbian Lite and then screenly-ose installed, then all updates done, like: 
`sudo update` 
`sudo apt full-upgrade`
`sudo rpi-update`

at this point since I was testing other things with libwebkit, I dont remember what else was installed, but if you reproduce my setup and it doesnt work we can start looking into what packages that deal with web libraries we dont have in common..

ealmonte32, 2019-03-28:
@jeffdesc 
This issue should be closed since it works now on production version as well as development.
I am able to reproduce a working asset from https://play.grafana.org/ every time on different Pis.

closed 2019-03-28 by vpetersson