Screenly/Anthias #819: Certain Wordpress sites are not showing in Screenly

micb2, 2018-05-15:
We have sites created by Wordpress with URLs in the following format:

https://domain.x.y.de/path/index.php/path/to/something/

In the Syslog it says that the page is showing but that is not the case:

`May 15 10:43:39 raspberrypi python[595]: update_playlist
May 15 10:43:39 raspberrypi python[595]: Generating asset-list...
May 15 10:43:39 raspberrypi python[595]: generate_asset_list deadline: 2009-04-17 14:56:00
May 15 10:43:39 raspberrypi python[595]: get_next_asset after refresh
May 15 10:43:39 raspberrypi python[595]: get_next_asset counter 0 returning asset 2 of 2
May 15 10:43:39 raspberrypi python[595]: /usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
May 15 10:43:39 raspberrypi python[595]:   InsecureRequestWarning)
May 15 10:43:39 raspberrypi python[595]: Showing asset https://domain.x.y.de/path/index.php/path/to/something/ (webpage)
May 15 10:43:39 raspberrypi python[595]: Asset URI https://domain.x.y.de/path/index.php/path/to/something/
May 15 10:43:39 raspberrypi python[595]: current url is https://domain.x.y.de/path/index.php/path/to/something/
May 15 10:43:39 raspberrypi python[595]: Sleeping for 30`

It is not an HTTPS issue as the situation is the same with HTTP.

On the big screen Screenly is still showing the previous asset which is just a plain URL like "https://aaa.bbb.de". Uploaded images are working as expected as well.

We are using the OSE version contained in the image 2018-05-04-Screenly_OSE_4GB.zip.

How can we solve or troubleshoot hat issue?

antonmolodykh, 2018-05-15:
Can you run `sudo journalctl > foobar.txt` and attach foobar.txt?

micb2, 2018-05-15:
Sure, thanks for your quick reply.

By the way, the Wordpress admin changed the link format to https://aaa.bbb.ccc.de/sermon?page_id=94 for testing and it is still not working.

`pi@raspberrypi:~ $ cat foobar.txt
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Tue 2018-05-15 11:04:33 CEST. --
Nov 03 18:16:42 raspberrypi kernel: Booting Linux on physical CPU 0x0
Nov 03 18:16:42 raspberrypi kernel: Linux version 4.14.34-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1110 SMP Mon Apr 16 15:18:51 BST 2018
Nov 03 18:16:42 raspberrypi kernel: CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Nov 03 18:16:42 raspberrypi kernel: CPU: div instructions available: patching division code
Nov 03 18:16:42 raspberrypi kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Nov 03 18:16:42 raspberrypi kernel: OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Nov 03 18:16:42 raspberrypi kernel: Memory policy: Data cache writealloc
Nov 03 18:16:42 raspberrypi kernel: cma: Reserved 8 MiB at 0x33400000
Nov 03 18:16:42 raspberrypi kernel: On node 0 totalpages: 211968
Nov 03 18:16:42 raspberrypi kernel: free_area_init_node: node 0, pgdat 80c84e40, node_mem_map b2caf000
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 1863 pages used for memmap
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 0 pages reserved
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 211968 pages, LIFO batch:31
Nov 03 18:16:42 raspberrypi kernel: random: fast init done
Nov 03 18:16:42 raspberrypi kernel: percpu: Embedded 17 pages/cpu @b2c58000 s38720 r8192 d22720 u69632
Nov 03 18:16:42 raspberrypi kernel: pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
Nov 03 18:16:42 raspberrypi kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Nov 03 18:16:42 raspberrypi kernel: Built 1 zonelists, mobility grouping on.  Total pages: 210105
Nov 03 18:16:42 raspberrypi kernel: Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbdepth=32 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=26a423b4-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash vt.global_cursor_default=0 quiet init=/lib/systemd/systemd net.ifnames=0
Nov 03 18:16:42 raspberrypi kernel: PID hash table entries: 4096 (order: 2, 16384 bytes)
Nov 03 18:16:42 raspberrypi kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Nov 03 18:16:42 raspberrypi kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Nov 03 18:16:42 raspberrypi kernel: Memory: 818436K/847872K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 21244K reserved, 8192K cma-reserved)
Nov 03 18:16:42 raspberrypi kernel: Virtual kernel memory layout:
                                        vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                        fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                        vmalloc : 0xb4000000 - 0xff800000   (1208 MB)
                                        lowmem  : 0x80000000 - 0xb3c00000   ( 828 MB)
                                        modules : 0x7f000000 - 0x80000000   (  16 MB)
                                          .text : 0x80008000 - 0x80800000   (8160 kB)
                                          .init : 0x80b00000 - 0x80c00000   (1024 kB)
                                          .data : 0x80c00000 - 0x80c8fd4c   ( 576 kB)
                                           .bss : 0x80c96f4c - 0x80d478b4   ( 707 kB)
Nov 03 18:16:42 raspberrypi kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Nov 03 18:16:42 raspberrypi kernel: ftrace: allocating 25231 entries in 74 pages
Nov 03 18:16:42 raspberrypi kernel: Hierarchical RCU implementation.
Nov 03 18:16:42 raspberrypi kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Nov 03 18:16:42 raspberrypi kernel: arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Nov 03 18:16:42 raspberrypi kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Nov 03 18:16:42 raspberrypi kernel: sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Nov 03 18:16:42 raspberrypi kernel: Switching to timer-based delay loop, resolution 52ns
Nov 03 18:16:42 raspberrypi kernel: Console: colour dummy device 80x30
Nov 03 18:16:42 raspberrypi kernel: console [tty1] enabled
Nov 03 18:16:42 raspberrypi kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Nov 03 18:16:42 raspberrypi kernel: pid_max: default: 32768 minimum: 301
Nov 03 18:16:42 raspberrypi kernel: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
Nov 03 18:16:42 raspberrypi kernel: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
Nov 03 18:16:42 raspberrypi kernel: Disabling memory control group subsystem
Nov 03 18:16:42 raspberrypi kernel: CPU: Testing write buffer coherency: ok
Nov 03 18:16:42 raspberrypi kernel: CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Nov 03 18:16:42 raspberrypi kernel: Setting up static identity map for 0x100000 - 0x10003c
Nov 03 18:16:42 raspberrypi kernel: Hierarchical SRCU implementation.
Nov 03 18:16:42 raspberrypi kernel: smp: Bringing up secondary CPUs ...
[... cut ...]
May 15 11:04:30 raspberrypi systemd[1419]: Reached target Basic System.
May 15 11:04:30 raspberrypi systemd[1419]: Reached target Default.
May 15 11:04:30 raspberrypi systemd[1419]: Startup finished in 61ms.
May 15 11:04:30 raspberrypi systemd[1]: Started User Manager for UID 1000.
May 15 11:04:33 raspberrypi sudo[1447]:       pi : TTY=pts/1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl
May 15 11:04:33 raspberrypi sudo[1447]: pam_unix(sudo:session): session opened for user root by pi(uid=0)`

micb2, 2018-05-15:
Sure. Sorry, I'll have to get used to GitHub.

[test.txt](https://github.com/Screenly/screenly-ose/files/2004234/test.txt)

antonmolodykh, 2018-05-15:
@micb2  I don't see any obvious problems in the logs. Can you give an example of any page with which this problem occurs?

micb2, 2018-05-15:
@antonmolodykh I'll talk to the administrator. Is there a way to trigger the download of the webpage manually from CLI in the way screenly would do it so that we can troubleshoot from there?

antonmolodykh, 2018-05-15:
OSE works a little differently.
if short, the viewer simply reads the assets from the database and shows it.
that's what I think: if I'm not mistaken and there really is not enough information in the logs, then maybe the problem is on the `uzbl` side.
you can try to check it yourself: for this stop the viewer (`sudo systemctl stop screenly-viewer`) and try to run the page in `uzbl` (don't forget to set the `DISPLAY=:0.0` environment variable)

micb2, 2018-05-15:
Thanks for explaining how to test it from shell. Attached you find the output.

[test2.txt](https://github.com/Screenly/screenly-ose/files/2004941/test2.txt)

It stucks after this line:

`External JavaScript file /home/pi/.local/share/uzbl:/usr/share/uzbl/examples/data:scripts/follow.js loaded`

There shouldn't be that heavy javascript operations involved as it is more or less basic Wordpress. After the Javascript part the page is showing up and looking fine on the screen. The loading time is way too long (> 2minutes, longer than the asset duration configured in Screenly).

vpetersson, 2018-05-15:
@micb2 what Raspberry Pi hardware version?

micb2, 2018-05-15:
@vpetersson It says Raspberry Pi 3 Model B V1.2 on the chip.

vpetersson, 2018-05-15:
Good. I just wanted to rule out the possibility that you were trying this on an old Raspberry Pi.

micb2, 2018-05-15:
Is there a way to speed up the javascript process or to find out why it is that slow or to skip it at all?

micb2, 2018-05-16:
We found out that the Wordpress is using a font that the Raspi wants to fetch from Google. The problem is that Screenly is not on the internet. Is it possible to pre-install the font somehow so that it is not necessary to fetch it from Google?

micb2, 2018-05-17:
We found a workaround by ourselves by adding a proxy.

For making Screenly work with the proxy, I had to make some changes, especially disable your Python "url_fails"-checkings:

`/home/pi/.config/uzbl/config-screenly
set proxy_url = http://w.x.y.z:1234

/home/pi/screenly/server.py
line 302
        # if url_fails(asset['uri']):
            # raise Exception("Could not retrieve file. Check the asset URL.")
line 441
        # if url_fails(asset['uri']):
            # raise Exception("Could not retrieve file. Check the asset URL.")

/home/pi/screenly/viewer.py
line 388
    # elif path.isfile(asset['uri']) or not url_fails(asset['uri']):
    elif path.isfile(asset['uri']) or True:
`

I'm sure that there are way more elegant ways to do this (feature request?).

vpetersson, 2018-06-21:
@micb2 You could try the experimental branch that @antonmolodykh is working on to see if that solves the problem.