Screenly/Anthias #821: Removing Resin-wifi-connect

rexlance, 2018-05-22:
Hi,

Is there a way i could remove resin-wifi-connect or stop the service?

Thanks.

antonmolodykh, 2018-05-22:
Hi @rexlance  to disable the service, you can run `sudo systemctl disable wifi-connect.service`.
can I ask the reason? something does not work correctly?

rexlance, 2018-05-22:
Hi @antonmolodykh thanks! Would try it out.
Everything is working fine. Just that we are using hostapd (access point) as we do not have wifi at our outlet.

Unless resin-wifi-connect is able to provide such services too? Thanks!

vpetersson, 2018-05-22:
@rexlance Please note that if you do not have an internet connection on your Raspberry Pi, you will need to add a Real Time Clock (RTC) device (since it lacks one). If not, the clock will not work.

rexlance, 2018-05-24:
@vpetersson Yes, we have been using RTC.

How do I set a dhcp/static ip address for screenly to read after that. Is it still in /etc/network/interfaces?

rexlance, 2018-05-24:
@antonmolodykh  after issuing sudo systemctl disable wifi-connect.service
added: 
"auto lo
iface lo inet loopback
iface eth0 inet dhcp
allow-hotplug wlan0
auto wlan0
iface wlan0 inet dhcp
        wpa-ssid "ssid"
        wpa-psk "password""

to /etc/network/interfaces and reboot.

Got a EXT4-fs error (device mmcblk0p2): ext4_iget:4697: inode #252929: com systemd-tmpfile: bad extra_isize 65535 (inode size 256)

on boot up. Any idea why?

Thanks.

vpetersson, 2018-05-24:
Sounds like a corrupt SD card. I'd suggest doing an Overwrite Format using [SD Formatter](https://www.sdcard.org/downloads/formatter_4/) and then re-flash with the latest disk image.

Also, keep in mind that if you use the WiFi connect to the WiFi, you need to delete the wifi profile too.

rexlance, 2018-05-25:
I did flash my sd cards. I have a few of them.
After issuing systemctl disable wifi-connect.service without changing other things. I rebooted into the same error.

vpetersson, 2018-05-28:
If you're still getting the `Got a EXT4-fs error (device mmcblk0p2): ext4_iget:4697: inode #252929: com systemd-tmpfile: bad extra_isize 65535 (inode size 256)` error on a different SD card, then it is most likely either your SD card reader that is broken (we've seen this surprisingly often) or the download is corrupt.

To rule out both, I'd recommend that you:
 * Try using a different SD cards
 * Validate the downloaded file with the MD5 checksum provided

vpetersson, 2018-08-13:
Closing out due to lack of response.

closed 2018-08-13 by vpetersson