Screenly/Anthias #823: URL with dynamic display crashes

bhagemann, 2018-05-24:
#### Overview of the Issue

Accessing our solar panel array's dashboard with the 'kiosk' URL displays a javascript driven dynamic page that changes every so often to show different stats.  If I have this in the list of assets the pi will eventually lock up and either show a blank screen, or this dashboard (and unresponsive to the admin page or ssh access).  recycling power fixes.  It will generally happen within 24 hrs.  I've seen it go a bit longer and lock up in a shorter time frame however.  If I remove the URL from the asset list, it runs day after day with no issue.

Also occasionally (every 4th or 5th time it shows) it will have trouble loading the URL.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* URL asset is:  https://monitoringpublic.solaredge.com/solaredge-web/p/kiosk?guid=4df18913-3506-4aea-9208-e576bb78003f
* have it display for about a minute or 2
* show some jpg slides for 30 sec each, then start over
* screenly stops working after a bit of time <24 hrs.  (its usually after I leave for the day but before returning)

### Environment

* Raspberry Pi 3, (I've tried diff. ram but currently 8gb Samsung card)
* Screenly OSE Version: currently using sprint 7, also tried updated to dev version

vpetersson, 2018-06-04:
@bhagemann Keep an eye open on the experimental release we are hopefully shipping this or next week which will include a new browser, which in turn will hopefully resolve this issue.

bhagemann, 2018-06-05:
Will keep an eye out for an announcement for that, will it be the same as updating via the bash shell command and accepting the development version?

vpetersson, 2018-06-22:
@bhagemann - The new release is now available. Run the upgrade script and select the experimental branch.

ping @antonmolodykh.

closed 2018-08-28 by stale[bot]