Screenly/Anthias #824: "Update Browser to View Embedded Content

greenie1987, 2018-05-29:
I'm running Screenly OSE on my Raspberry Pi 3. I created a public website using Google Slides to display local weather information in additional to other local items on the page. The page displays on my monitor however I get an error message that says "Update your browser to view embedded content". 

Is it possible to update my browser if so, does anyone have any ideas on how to accomplish this? 

Thanks!

vpetersson, 2018-05-29:
See #527. In short, the browser the current version of Screenly is using is getting out of date so we are working at replacing it.

antonmolodykh, 2018-06-20:
@greenie1987 we are happy to announce that the experimental Screenly OSE version is available. It includes the new browser with hardware acceleration support (as a replacement for old uzbl), and migration of all services to the docker. To try it follow the steps under 'Manual upgrade' [here](https://www.screenly.io/ose/) and answer yes to the question about experimental version.
We are looking forward to feedback, please let us know about any problems.