Screenly/Anthias #827: Cannot upload large files.

Zorbeen98, 2018-06-05:
#### Overview of the Issue

I have a 1.04GB video that I'd like to add as an asset. 
Whenever I try to upload it it runs for a while and then I get 'The operation failed. Please reload the page and try again.' 

I've tried using IE and Chrome. 

I tried adding 'client_max_body_size 50000000m;' to the nginx.conf settings but it doesn't seem to have helped. 

Found old issue here but it has reportedly been resolved...
https://github.com/Screenly/screenly-ose/issues/444

#### Reproduction Steps

Upload large mp4 file through web page.

### Environment

* Raspberry Pi Hardware Version: Pi 3 B+
* Screenly OSE Version: Not sure how to tell... (04/05/2018)

Zorbeen98, 2018-06-07:
The issue appears to be caused by the wifi disconnecting when uploading a large file. 

Not sure if anything can actually be done to resolve the the issue or whether it's just down to the NIC on the Pi...

vpetersson, 2018-06-22:
Just a random idea - What power supply are you using? I know that if you are using an under-powered power supply, you can see strange network issues during high load.

closed 2018-08-10 by vpetersson