Screenly/Anthias #831: Content Scheduling

fabioulous, 2018-06-12:
Hi! I recently discovered screenly ose and it is amazing, exactly what I was looking for. Except only one crucial feature is missing:
Content scheduling, for example: Display a breakfast menu every morning and a dinner menu in the evening.
If you could implement this feature I would be most thankful! 
best wishes
Fabi

vpetersson, 2018-06-12:
@fabioulous This is something we already support in Screenly Pro. See [this article](https://support.screenly.io/hc/en-us/articles/217874263-Using-Screenly-s-sophisticated-playlist-editor).

fabioulous, 2018-06-14:
@vpetersson Thank you for the hint. My company is not looking for a cloud based solution. We want to have the management software running on our servers as well as the content locally stored.

antonmolodykh, 2018-06-14:
@fabioulous to display assets at a specific time you can use API (`http://raspberry_ip/api/docs/`). I think a script will be pretty simple.
If you go this way, I recommend you to update Screenly from the development branch (we improved API a couple of days ago).

closed 2018-10-19 by stale[bot]