Screenly/Anthias #836: Fix usage on Pi Zero

vpetersson, 2018-06-21:
#### Overview of the Issue

There appears to be two separate (but perhaps related) issues when using the latest disk image on the Pi Zero:
 * The splash screen is not working
 * When X11 is loaded up, it just shows a black screen, which means we are unable to configure WiFi

#### Reproduction Steps

* Flash out the latest disk image (2018-05-04).
* Boot the system
* The screen will just show a black screen at the end of the boot

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi Zero
* Screenly OSE Version: 2018-05-04 disk image

vpetersson, 2018-06-22:
I can confirm that the new experimental branch does not work on Pi Zero either.

antonmolodykh, 2018-07-06:
I think this report is partially incorrect.
Pay attention to [it](https://github.com/Screenly/screenly-ose/blob/master/start_resin_wifi.py#L30). in fact hotspot page is generated before `wifi-connect` starts. The page should have been displayed in any case if there is no connection. There is a problem: script does not consider devices without wifi (fixed in a [last PR](https://github.com/Screenly/screenly-ose/pull/849)).
 
Pi zero w (I tested on zero w) shows a black screen for about 20 seconds, but then the viewer starts up. I think this is due to low productivity of Pi zero. In addition, starting a device without a network connection is always slower than starting a device with a network connection.

The splash screen is sometimes not displayed (but sometimes it does) on pi zero and pi 1, I think  this is also related to low productivity devices and I'm not sure we can fix this.

antonmolodykh, 2018-07-06:
However, I confirm that the experimental version does not work on pi zero w for now.

closed 2018-10-22 by vpetersson